gusmanb/logicanalyzer #2: Cross Platform Software

DaClemens, 2022-07-04:
First of all, cool work!

You said you might make a cross platform UI if you get enough requests. I haven't seen such a request to +1 it yet, so I'll start a first one.

Personally I'll receive my Pico in the next couple days and I want to try this out immediately, even if it's on Windows only. I'd still prefer a Linux version.

Thanks for the good work!

hjhjdev, 2022-07-05:
Nice work!! I'd love to see this app in macOS.

gusmanb, 2022-07-05:
Wow, didn't expected such a fast spreading of the project... 😄 

Ok, I will start this weekend to test .net MAUI and if it's mature enough I will port it to mac/linux/iOS/Android (I'm planning a 16 channel wireless version with the pico W and using it with a tablet or phone can be very useful).

Will leave this issue open to add updates about the portable version.

caljorden, 2022-07-05:
I would suggest you might investigate the possibility of leveraging the excellent [Sigrok](https://sigrok.org/) open source project (along with it's primary GUI, pulseview) as the cross-platform software you use for this project.  It has a nice plugin architecture for adding drivers for new devices, and supports decoding numerous protocols.

gusmanb, 2022-07-05:
> I would suggest you might investigate the possibility of leveraging the excellent [Sigrok](https://sigrok.org/) open source project (along with it's primary GUI, pulseview) as the cross-platform [...]

I had many problems with Sigrok (I have updated the readme with onfo about it) and that's why I created my own app, in any case I will give it another try.

dhennessy, 2022-07-05:
+1 to request for sigrok driver

MacroController, 2022-07-07:
Would it be useful to make another issue thread for the Sigrock feature request, just to make it easier to follow and differentiate the traction for cross-platform availability, and the traction for the Sigrock integration feature?

X-Ryl669, 2022-07-07:
For sigrok users, there's [this implementation for RP2040](https://github.com/pico-coder/sigrok-pico). 

This repository contains everything in one place and it's very well explained, so if the software were running on Linux, it'd be perfect to use. Sigrok is more complex to set up and use, not sure about the linked driver.

janhendrik-rust, 2022-07-07:
Thanks for sharing this awesome project. There is so much to learn from it already.
Id' like to add my +1 for a cross platform UI.

gusmanb, 2022-07-07:
> Would it be useful to make another issue thread for the Sigrock feature request, just to make it easier to follow and differentiate the traction for cross-platform availability, and the traction for the Sigrock integration feature?

New issue has been created for Sigrok driver feature.

#4

DatanoiseTV, 2022-07-07:
> > @gusmanb As for cross platform gui, you should take a look at dear imgui. It's an excellent GUI library with great performance. Check this thread for some examples: [epezent/implot#180](https://github.com/epezent/implot/discussions/180)
> 
> I will check if it supports iOS, Android, Windows and macOS. I want to create a second version of the analyzer which uses a Pico W and has a LiPo battery, it will be limited to 16 channels but as it will be fully wireless it will allow to connect the analyzer to a phone or tablet, so you will not need a computer near the analyzer (very handy for restricted size work areas or even testing in the field).

Yes, check the backends and example folder at https://github.com/ocornut/imgui

gusmanb, 2022-07-09:
After checking many UI frameworks for .net I have decided which one I'm going to use. 

I have checked:

- .net MAUI
- ImGUI.net
- wxWidgets
- Qt
- Blazor
- Electron.net
- Avalonia UI

My initial idea was to use .net MAUI, but the Linux support is inexistent. There is a community project that adds support for it but it does not seem to be very active so I discarded this option.

ImGUI, wxWidgets and Qt present for me the same problem: their way of working is so different to what I'm used to that it would take me too much time to have anything usable.

Blazor and Electron are two options that I have considered heavily, I use Blazor in my job and creating a self-hosted app is something very viable, I'm comfortable with HTML and CSS so these two are very good options.

But at the end I decided to use Avalonia UI, it has a very big community backing it, I like the way it's implemented (very similar to Xamarin.Forms or .net MAUI), it's well documented and it has support for live preview in VS2022 what makes a lot easier for me to develop the app. I think it's time to dust off my XAML programming skills 😄 

I will update this issue as I make progress.

Cheers.

gusmanb, 2022-07-10:
This starts to look promising.

[image]

It takes a lot more time to create the GUI than with WinForm but I like a lot the aspect that it's getting 😄 

If everything continues at this speed I think next weekend I will have a first version to test.

gusmanb, 2022-07-18:
More news on the multiplatform app.

https://github.com/gusmanb/logicanalyzer/discussions/15 Multiplatform application

gusmanb, 2022-07-30:
Cross-platform application has been released. [Release 2..0 discussion](https://github.com/gusmanb/logicanalyzer/discussions/18)

closed 2022-07-30 by gusmanb