gusmanb/logicanalyzer #5: New features for the "level adapter" board: more logic levels 2.5, 1.8 and eventually reversed as signal generator..

aventuri, 2022-07-07:
Hello, that's a very nice project, this one!
i did a very clever usage of that "small" FSM present in these PIO modules.. didn't know of them, but very powerful, this way.
i'll not chime about the high level software, as it's open source and anybody can adapt to his best pet frontend! :-)

i'm more interested about the board for adapting different logic levels. i see the TXU0104 is pretty cool too, because it can manage also "external levels" from 1.1 to 5.5v so i suppose that many would be interested in a logic analyzer for also the "quite common" logic voltage of 1.8v and 2.5v. some simple extension would give this adapter board a quite more "niche" market, imho.

on the other hand, reading the DS of this TXU0104, it looks like it could also work in reverse mode, so you could use it also to have that good as a simple signal generator, for some of these pins. don't know if the state machine of these PIO units can be helpful for this task too, but in a general way, a device that can do both input and output could be pretty useful, for the same BOM or so.

that's all, sorry for the "noise", just my two cents

that's a good project already, anyway!

gusmanb, 2022-07-07:
Hi.

First of all, thanks for the comment, all the ideas are always welcome :)

About the TXU0104, this is the unidirectional family, the bidirectional one is the TXB*, it could be changed for the TXB ones but for the initial design did not made sense (also, the TXB has some other limitations, per example it cannot do up translation and down translation at the same time, what you use as VCCA is the minimum for VCCB).

In any case, the mod to include different voltages is possible and is one that I have planned for a future (not sure when as the first to do is the Linux and macOS app version, I'm a single developer and I'm creating this in my spare time😄) and not only fixed voltages but any voltage between 1.65 to 5.5v. Instead of powering the input side of the translators with a voltage coming from the analyzer is as simple as to power them from the device that you're going to capture from, but for this to be safe I need to add a protection circuit for over-voltages and an indicator of under-voltage.

When I receive the first boards that I have ordered and checked they work as expected I will try with 1.8v, the translators can reach up to 200Mbps but only when input is 3.3v and output is 5v, translating from 5v to 3.3v per example reduces the performance to 150Mbps, more than enough for the device, but when translating from 1.8v to 3.3v it reaches only 100Mbps and as it's on the limit, before implementing anything I want to test that it does not cause problems.

terenceyhj, 2022-07-14:
Hello,
Can we replace SN74lvc8t245(or SN74lvch16t245) with TXU0104 ? The cost will be lower!!

gusmanb, 2022-07-14:
Hmmm, are you sure it's cheaper? I buy the components in Mouser and right now the TXU0104 is 0.82€ while the SN74lvc8t245 is 1.95€, in total the TXU are 4.92€ and the SNs would be 5.85€...

bluegizmo83, 2022-08-22:
> Hi.
>
> First of all, thanks for the comment, all the ideas are always welcome :)
>
> About the TXU0104, this is the unidirectional family, the bidirectional one is the TXB*, it could be changed for the [...]

I'm thinking about ordering the PCB's for this Pico logic analyzer board, but before I do I'm curious if you had started working on the new boards with the wider voltage support levels? I would hate to order the current boards and parts and then have to reorder the updated boards if your already working on the update...? I thought I read somewhere too that you might integrate the logic level shifting onto the same PCB instead of it being a separate board once you switch to the wider voltage support, is that true?

gusmanb, 2022-08-23:
> I'm thinking about ordering the PCB's for this Pico logic analyzer board, but before I do I'm curious if you had started working on the new boards with the wider voltage support levels? I would hate [...]

Hi. About the shifter, yes, I have a new version but I haven't uploaded it, the change is minimal, I added a jumper that disconnects the +5V rail from the analyzer side connector, so the +5V pins of the input connector becomes ExtVREF, with this you can supply the reference voltage from the device you are going to measure and will work with any reference from 5.5v to 1.6v. I will try to upload it the next week as I'm also finalyzing the functionality to allow to stop a capture and I want to upload all at once.

About unifying all in a single board, it's a possibility but nothing that I have planned to do right now.

gusmanb, 2022-08-27:
Level shifter board has been updated, now you can choose to use the internal +5v as voltage reference for the input signals or you can supply the voltage reference from an external source (e.g. the device you're going to analyze). This provides to the board the ability to work with signals ranging from 1.65v to 5.5v.

closed 2022-08-27 by gusmanb