gusmanb/logicanalyzer #10: Resize Level Shifter PCB?

bluegizmo83, 2022-07-12:
Not an issue, just an idea:

Can you resize the Level Shifter PCB so that it's the same size as the main Logic Analyzer PCB and add holes to the four corners to align with the holes in the Analyzer PCB?

The idea is to mount the level shifter PCB upside-down on top of, or underneath of, the main Logic Analyzer PCB to keep the same footprint in the X and Y directions, and it would just end up being slightly taller in the Z direction.

I really don't know how to use PCB design software (i.e. KiCad) so I'm lost when trying to do it myself.

gusmanb, 2022-07-13:
Hi.

Implementing what you expect is not as simple as you think. 

First, the device uses horizontal pin headers as inputs, if you replace that header with straight pins at the bottom of the analyzer then using it without the shifter board would be a pain, to be usable the main board would need to be modified with a second header and it would cause the device itself to be bigger and would make impossible to place the device on a table/bench/whatever stably as the connectors would be touching it. Also, making a housing for the device would become a lot more difficult. Finally I think it would feel strange that when you use the device without the shifter you will connect the wires to the left but with the shifter to the right with all the pins mirrored.

The device itself when assembled with the shifter is small, only 10.5x6.5cm, so the space gain is not noticeable. In any case I have planned in the future to modify the shifter to include multi-voltage inputs, and if it works correctly then create a final version of the analyzer where the shifter board is integrated in it, but for now the main goal is to have the option to assemble what you need as cheaper as possible.

Cheers.

closed 2022-07-13 by gusmanb

bluegizmo83, 2022-07-13:
All good points, I should've thought of that! Thanks for the reply though.