gusmanb/logicanalyzer #11: How to stop capturing?

superyawei, 2022-07-13:
Hi, brother
When I am capturing, I want to stop capturing; But I don't know how to stop scanning, only power off and restart.
Adding start and stop buttons to the software is a good choice.

gusmanb, 2022-07-13:
Hi.

Thanks for the suggestion.

Right now there is no way to stop the capture becausue when I was developing the device I had some problems reading from the USB while the capture was running at full speed, so USB comms are not checked when the capture is running. I think the bus matrix is so saturated that it makes the RP2040 to crash.

In any case I will check it again and if I see that it works without causing problems will add the function to stop the capture.

Cheers.

gusmanb, 2022-08-27:
Functionality to stop a running capture is under testing, probably next week will be published.

superyawei, 2022-08-27:
very much looking forward to

gusmanb, 2022-09-03:
Implemented in [release 2.0.1.0 ](https://github.com/gusmanb/logicanalyzer/releases/tag/v2.0.1.0)

closed 2022-09-03 by gusmanb