gusmanb/logicanalyzer #17: Some questions about samples deep

vihor3, 2022-07-27:
Sorry, first of all, please forgive me for asking some novice questions.
What bottleneck limits the sample depth? Is it CPU computing power, memory size, or something else?

gusmanb, 2022-07-27:
What do you refer to with "bottleneck"? Sampling speed?

If you mean what limits the maximum number of samples, it's memory, unfortunatelly the RPico only has 264Kb of RAM and external RAM is not an option, the QPI bus is used by the flash and regular SPI memories are too slow for this application.

If you mean the sampling speed, it's the CPU max frequency, the program is extremely optimized, uses two PIO ASM instructions per capture so unless you overclock (more) the pico it's not possible to achieve more speed, but in any case it's already faster than any other analyzer that only uses a microcontroller for the task, other analyzers use expensive FPGA's to achieve similar results.

vihor3, 2022-07-27:
Then I think the CPU can be slightly overclocked. As far as I know, overclocking to more than 250M should be very easy. I saw that an oscilloscope project had been overclocked to 280M and operated stably. According to the author, his chip was still inferior in quality

vihor3, 2022-07-27:
This is the link of rp2040 oscilloscope. https://github.com/panyihang/Cube-436_miniOSC

gusmanb, 2022-07-27:
Nice project :)

That oscilloscope is using a custom power supply overvoltaging the RP2040, that's why it can reach 280Mhz. That has multiple disadvantages: you cannot use any more a pico, you need to create a full custom board (and in that case instead of a RP2040 you could use something like a STM32H750 which reaches 480Mhz and has 1Mb of RAM), makes a lot less accessible the project to people that does not have advanced SMD soldering skills, not all the picos may reach that frequency and finally you are reducing significantly the life span of the RP2040.

I already overclock the pico to 200Mhz, I have 6 picos from different batches and all work ok at this frequency, from the six I have only one supports 250Mhz with the base voltage, others reach up to 224Mhz, but all work at 200Mhz, that's why I choosed this frequency.

closed 2022-07-28 by gusmanb