gusmanb/logicanalyzer #20: Channel name and GPIO number don't match

X-Ryl669, 2022-09-01:
Thanks for the tool, it works very well.
However, I've an issue with the reported channel number and the board GPIO number.
Typically, I'm plugging a I2C connector with (VCC, GND, SCL, SDA) pin on GPIO (5, GND, 6, 7) (VCC is just because I already have a 4 pin connector and I don't want to make an adapter).

In the interface, I get SCL on Channel 5, SDA on Channel 6 and zero on channel 7. I've also tried to export the capture and it's labeled the same way. Please notice that it's not a real issue since I can identify the signals from their shape, but it's disappointing.

I've tried to plug the connector on GPIO (18, GND, 17, 16) and in that case, I capture SCL on Channel 17, but I'm not successful in capturing SDA on either channel 16 or 18 (both are zero).

gusmanb, 2022-09-01:
Hi.

Channels and GPIO's don't use the same numeration, if you check the schematic you will see that GPIO2 is the first channel as GPIO0 and GPIO1 are used by the system for the PIO trigger machine.

[image]

Also, something that I did that is not totally correct is that I shifted the channel names by one in the software, to make them more easily understandable, so what in the hardware is labeled "Channel_0" in the software is "Channel 1".

About labeling, remember that you can set any name to any channel that you want from the GUI, each channel has a textbox below the channel number and if you set there the signal name it will be exported with that name.

[image]

Cheers

X-Ryl669, 2022-09-01:
Ok, that makes sense. 
Maybe a legend could be useful in the software, something like:

| GPIO | Channel number |
|-|-|
|0 & 1 | N/A |
| 2 | 1 |
| 3 | 2 |
| 4 | 3 |
| 5 | 4 |
| 6 | 5 |
| 7 | 6 |
| 8 | 7 |
| 9 | 8 |
| 10 | 9 |
| 11 | 10 |
| 12 | 11 |
| 13 | 12 |
| 14 | 13 |
| 15 | 14 |
| 16 | 15 |
| 17 | 16 |
| 18 | 17 |
| 19 | 18 |
| 20 | 19 |
| 21 | 20 |
| 22 | 21 |
| 26 | 22 |
| 27 | 23 |
| 28 | 24 |

gusmanb, 2023-02-25:
Closed as this is now documented in the [Hardware section of the Wiki](https://github.com/gusmanb/logicanalyzer/wiki/02---LogicAnalyzer-Hardware).

closed 2023-02-25 by gusmanb