gusmanb/logicanalyzer #21: Ability to delete useless samples.

X-Ryl669, 2022-09-01:
In the logic analyser if would be very useful if we could select a time range where we don't care about the signal and hit delete to remove it. It way much easier than scrolling when event are far apart.

Also, can you increase the minimum zoom level (even with 200 samples in view, it's still too much zoomed IMHO, I'd like to see more signals)

Thanks!

gusmanb, 2022-09-01:
I like the idea of deleting a range of unneeded samples, I will add this to the next development round :)

About the zoom, yes, I can increment as much as we want but I'm worried about performance, one of the goals of the device is to have something very cheap that can work in machines that aren't very powerful and increasing the number of samples in screen in something like a raspberry 3 can make it to be so slow that you will need to kill the application. I will think something to warn the users before going beyond 200 samples in screen and will then update thee range up to 500 samples.

Cheers!

gusmanb, 2023-01-29:
New release now includes the ability to delete samples, check the main page to know how to use it.

closed 2023-01-29 by gusmanb