gusmanb/logicanalyzer #22: How to use software

PersistentBash, 2022-09-10:
Where and what files for windows users need to use/install the software?

gusmanb, 2022-09-10:
In the [releases](https://github.com/gusmanb/logicanalyzer/releases) section download the zip file for win-x64, uncompress and it will be ready to use.

closed 2022-09-10 by gusmanb

Gustavomurta, 2022-11-20:
Dr. Gusman, your project is fantastic and it works! I've done some tests and managed to capture digital signals. I will do more specific tests. Thank you so much for sharing your project.
I would like to suggest the insertion of the link of the compiled programs in the main page. As I didn't know about that link, I had to compile on Windows 10 with Visual Studio 2022. And it compiled OK.

gusmanb, 2022-11-20:
> Dr. Gusman, your project is fantastic and it works! I've done some tests and managed to capture digital signals. I will do more specific tests. Thank you so much for sharing your project. I would [...]

Thanks, glad it compiled correctly.
This is how Github works, you can check that nearly any project here has the exact same structure, you will find always a link to the releases (if the project has any) in the main page and there you will find all the compiled files.

Also, sorry but I have deleted your other issue, I only allow links to projects in github, in other case the comments get cluttered with people spamming to promote their own blogs and sites. If you publish the project in Github feel free to open a new issue linking to the repository and I will leave it there.

In any case, this device does not need calibration at all, that's the beauty of the PIO units, it's deterministic, each instruction takes always a single cycle (unless you specify a delay for it, another of its wonders) so just knowing the base frequency and the instruction count you get absolute precission on the time base (I verified it with my [200 Mhz clock generator](https://github.com/gusmanb/multiosc) and a Rigol oscilloscope).

Gustavomurta, 2022-11-21:
Well, Dr, I didn't find any link to the releases on the main page of the project. If the intention is to publicize the project, I still suggest that you disclose the link. For example, I found the link by chance in one of the project's issues.
As I hadn't found the compiled programs, I followed your suggestion - I installed Visual Studio 2022 and compiled the solution in C#. Note: I had never compiled in C#, but everything went well. Thanks.
About the ESP32 frequencymeter project, I shared the blog link, because that is where the complete project documentation exists. I'm sending the link from Github. As soon as possible, I will translate the documentation.
https://github.com/Gustavomurta/ESP32_frequencyMeter/tree/master/ESP32%20FreqMeter%20Arduino

gusmanb, 2022-11-21:
Hi Gustavo.

I have added another link to the releases section in the main page, hope it's now more visible.

By default Github creates this link in the main page, you can check it in another repositories:
[image]

I agree that for someone that is not very used to Github it may be a bit difficult to find it so I hope the new link makes it easier.

Also, I have added a [thread in the discussion section with related tools](https://github.com/gusmanb/logicanalyzer/discussions/30), I have added your frequency meter, it will be more visible as this issue is closed and it is not usually shown. Feel free to add any other tool that is related.

Regarding colaboration, I have left a mesage in your repo issues, if you're interested I would suggest to create a new thread in the [discusions section](https://github.com/gusmanb/logicanalyzer/discussions) so we can chat there and if anyone else wants to colaborate can join it.

Cheers!

Gustavomurta, 2022-11-21:
Thank you for your good intention. Thank you very much

Oliv4945, 2022-11-26:
Hello @gusmanb , thank you for this great software!
I wonder if I am using correctly the software (v2.0.1.0 - Linux Debian 11): when I click on `protocol analyzer` I only see `None`: 
[image]

Is there something I am doing wrong?

gusmanb, 2022-11-26:
> Hello @gusmanb , thank you for this great software! I wonder if I am using correctly the software (v2.0.1.0 - Linux Debian 11): when I click on `protocol analyzer` I only see `None`: [image]
> 
> Is there something I am doing wrong? Thanks

Doh... I just checked the packages and forgot to add the analyzers folder... If you copy it from the release 2.0 it should work, I will create a new package as soon as I have enough time.

Oliv4945, 2022-11-26:
Thanks, it does work :)