gusmanb/logicanalyzer #25: Option to make windows background solid instead of transparent

RSDuck, 2022-11-01:
Currently the logic analyzer window is transparent which I personally find distracting and to sometimes result in a bad contrast depending on the order of the windows behind it . It can also be noticably resource intensive.

For this reason I'd much appreciate if it was possible to set the window to be made of solid colors.

gusmanb, 2022-11-02:
Hi.

Sure, I will add an option to make the window solid. Possibly will have the version ready for this weekend.

Cheers.

gusmanb, 2022-11-26:
Hi.

I finally changed the background without option for transparency, it works better and it reads better so it has been a no brainer.

The code is already uploaded, in the next release it will be included.

Cheers.

closed 2022-11-26 by gusmanb