gusmanb/logicanalyzer #27: Pico-W support

revell1, 2022-11-07:
The current code hard codes the LED pin.
Would it be possible to add a build for the Pico-W, as the LED uses an IO pin on the Wifi chip.
Would not want/expect Wifi support, but would be useful to have a Pico-W build image for those of us who have spare Pico-W's.

gusmanb, 2022-11-07:
Hi!

In fact, one of the things that I have planned is to add wifi support for the W, I have one sitting in the bench only for that, but lately I've been busy between my job and other project that I'm working on.

Not sure when but the support for the W will come, that's for sure.

Cheers.

revell1, 2022-11-07:
Good to hear.
I would note that from current playing with Pico-W (and Arduino IDE toolchain) the Wifi functionality appears to be a bit unstable, also USB interrupts interact with Wifi, and TCP traffic is often slow/delayed.
A simple web server app I have written can often take a second to reply, while at other times it may only be 200-300ms or faster.
So not sure how that may impact on you data transfers at high speeds.

Just thought I would mention it in passing for future ref, as I think there is probably a lot in common between the Pico SDK and the Arduino IDE SDK.

gusmanb, 2022-11-07:
Thanks for the info :)

The W version will not use USB at all, it will be completely wireless, so for that side we're good.
Also, one of the particularities of this analyzer is that does not require to send data at high speeds to the host, this could even work in a 300bps serial port without affecting the performance of the device (that's why it's restricted to 32k samples, because all is done in memory, triggering and capturing).

revell1, 2022-11-07:
Just a note (from my observations to date), you would need to use an external USB PSU, as simply plugging the Pico-W into a USB port on a PC appears to cause enough random USB activity from the PC to cause issues, even if you are not using the USB connection for any data transfer. I am not sure what the PC is doing, but I guess there is some random interrogation that occurs,  or possibly it is some form of handshaking from the Pico device.

So the end user would probably need to be made aware that they need to use a standalone USB PSU to avoid potential issues.

gusmanb, 2022-11-07:
That's normal, when a device is pluged to USB it starts the detect stage where devices identify themselves to the host and once it is detected it may sometimes send a packet to test the device status. Disabling the USB controller using the register USB_MAIN_CTRL should totally stop this.

Hasenpups, 2023-01-16:
Is it possible to compile the firmware and use a Pi Pico W as a "Pi Pico without W". I only own a Pi Pico W and want to test a little bit with out firmware. I don't need the wifi part and want to use it via USB.

gusmanb, 2023-01-16:
It's not tested but it must work, the only difference between the pico and the pico w are some IO's that were not exposed now are used for the module, and the LED is now connected to the WiFi module instead to the RP2040, for the rest is the same board, you will not see the LED illuminated nor blinking but it will work.

I recommend you to comment the GPIO initialization and blink code (is all in the main in the C file) and you can try to use the W integrated functions to blink the led through the WiFi module, there should be enough room for it.

You can use as reference the picow led blink example: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/blink/

Cheers!

gusmanb, 2023-01-31:
> Just a note (from my observations to date), you would need to use an external USB PSU, as simply plugging the Pico-W into a USB port on a PC appears to cause enough random USB activity from the PC [...]

Hi.

I am implementing the WiFi support for the W (in fact I already have the firmware up and running and the command line tool working, I only need to finish the GUI app so it's very possible that today or tomorrow I will release it) and I have been testing the USB problem you commented.

At a first I tried to disable the USB whenever a TCP connection was made and re-enable it when it gets disconnected, but restarting the USB controller does not work, it does not redo the handshake with the OS and the USB port gets unusable.

I have been conducting tests and I haven't faced any problems having the TCP connection open and the USB connected to a PC, what was exactly the problem you faced? It dropped the network connections? Did the CYW device become unstable? Any more detailed info would be great.

I must say that I use sepparated cores for USB and WiFi, the core0 runs the main software and the USB handling and core1 is used exclusively to control the CYW device, that may be why I'm not having any problem as the interrupts are associated to a core and that may avoid the interferences between these.

Cheers.

gusmanb, 2023-01-31:
Support for the Pico-W has been added and WiFi connection is now possible.

closed 2023-01-31 by gusmanb