gusmanb/logicanalyzer #31: Crash when exporting data (Linux) - Index was outside the bounds of the array.

Oliv4945, 2022-11-26:
Hello,

I am using v2.0.1.0 under Linux, the capture and save are fine, but exporting the data results in an application crash with the following log:

```
Unhandled exception. System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at LogicAnalyzer.MainWindow.<MnuExport_Click>b__9_0(Int32 c) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\MainWindow.axaml.cs:line 72
   at System.Linq.Enumerable.SelectArrayIterator`2.ToArray()
   at LogicAnalyzer.MainWindow.MnuExport_Click(Object sender, RoutedEventArgs e) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\MainWindow.axaml.cs:line 72
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.CheckSignaled() in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 164
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 244
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at LogicAnalyzer.Program.Main(String[] args) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\Program.cs:line 14
Aborted
```

The file is 0 bytes

Capture parameters:
`̀``
{
   "Frequency":4000000,
   "PreTriggerSamples":5,
   "PostTriggerSamples":1024,
   "CaptureChannels":[
      20,
      21,
      22,
      23
   ],
   "TriggerType":0,
   "TriggerChannel":22,
   "TriggerInverted":true,
   "TriggerBitCount":0,
   "TriggerPattern":0
}
``̀`

gusmanb, 2022-11-26:
Hi.

Thanks for the report, I think I have found the problem. Could you please test this build? I don't have a Linux machine at hand to test it. This should correct the problem and disable the transparency (on supported OS's) to improve performance.

[linux-x64.zip](https://github.com/gusmanb/logicanalyzer/files/10097082/linux-x64.zip)

Cheers.

Oliv4945, 2022-11-26:
Thank you, your fix is working!

In the mean time I wrote a small ".lac" converter in Python to transform my data... It took me more time than I want to admit to find out that the data are saved in decimal :)

closed 2022-11-26 by gusmanb