gusmanb/logicanalyzer #34: LogicAnalyzer software crashing on Win10

thisisgregm, 2022-11-29:
LogicAnalyzer software crashing on Win10.

Latest win64 version 2.1.0.0.

Steps to replicate:
- Open Logic Analyzer software
- Open serial device, it reports itself as LOGIC_ANALYZER_V1_0
- Click Capture
- Choose any settings, for example capture CH01, trigger CH01
- Press Accept
- it waits a few seconds then crash

Tried to open it from a command line to see some errors, but nothing is reported.

How can I debug it and see some errors please?

Not 100% sure if it's an app or my permissions in Windows or a space in a program path ("C:\Program Files\LogicAnalyzer").
Can someone please confirm if they have the same problem? If not I'll look elsewhere (Windows)

gusmanb, 2022-11-29:
Hi.

I have tested with the latest uploaded version of the software and firmware using a path with spaces and I'm unable to reproduce the problem.

You can check the windows event log, there should be some info about the problem.
Also, if you have VS2022 the best is to download the source code and run it directly from Visual Studio, that will show you exactly where the error happens.

Do the crash give you any information or it simply closes the application?

thisisgregm, 2022-11-29:
It just closes by itself with no info given.

At work I think we have Panda antivir running. I had to ask IT admin to whitelist dll libraries to make the software running as I had errors reported by Panda. Now no reports from Panda, so I don't think the problem is here.

Will take Pico home and try it on a home PC.

I'll report back, but I guess it would be too obvious it it happens due to a software bug. Nobody reported similar issue before, Seems like it's my Windows. Will try to see the error log to check if it's anything there.

thisisgregm, 2022-11-30:
checked windows Log and have some weird entries there:
```
.NET Runtime Error
Application: LogicAnalyzer.exe
CoreCLR Version: 6.0.822.36306
.NET Version: 6.0.8
Description: The process was terminated due to an unhandled exception.
Exception Info: System.UnauthorizedAccessException: Access to the path 'C:\Program Files\LogicAnalyzer\captureSettings.json' is denied.
   at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategy(FileStream fileStream, String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize)
   at System.IO.StreamWriter.ValidateArgsAndOpenPath(String path, Boolean append, Encoding encoding, Int32 bufferSize)
   at System.IO.File.WriteAllText(String path, String contents)
   at LogicAnalyzer.Dialogs.CaptureDialog.btnAccept_Click(Object sender, RoutedEventArgs e) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\Dialogs\CaptureDialog.axaml.cs:line 226
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at LogicAnalyzer.Program.Main(String[] args) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\Program.cs:line 14
```

Access to the path is denied sorted by running it as an administrator (that's probably our work windows policy, will speak to the IT admin). However the path from other lines **C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\** is definitely not correct (I don't have **geniw** user on my computer).

Other entries in the error log are:

Application error:
```
Faulting application name: LogicAnalyzer.exe, version: 1.0.0.0, time stamp: 0x62cf0990
Faulting module name: KERNELBASE.dll, version: 10.0.19041.2193, time stamp: 0x7f7062e1
Exception code: 0xe0434352
Fault offset: 0x000000000002cd29
Faulting process id: 0x99c
Faulting application start time: 0x01d904b2e0b9176a
Faulting application path: C:\Program Files\LogicAnalyzer\LogicAnalyzer.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: e3180201-6422-4711-a0f6-5cb8b87afdce
Faulting package full name: 
Faulting package-relative application ID: 
```

Windows Error reporting information:
```
Fault bucket 2144646143600215964, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: LogicAnalyzer.exe
P2: 1.0.0.0
P3: 62cf0990
P4: KERNELBASE.dll
P5: 10.0.19041.2193
P6: 7f7062e1
P7: e0434352
P8: 000000000002cd29
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERB8B3.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERB98F.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERB99F.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERB99D.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERB9AE.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_LogicAnalyzer.ex_84799016b7ba40f29131f97a1882331e151550cd_6a8fbb2c_5ec67363-752b-4915-9eaf-2366dd2509d1

Analysis symbol: 
Rechecking for solution: 0
Report Id: e3180201-6422-4711-a0f6-5cb8b87afdce
Report Status: 268435456
Hashed bucket: 31f13d43dd04b4c5edc3504e0633c79c
Cab Guid: 0
```

As I said managed to run it as admin with no errors, so please confirm this is my OS problem (if you think it is) and I'll close the issue here.

gusmanb, 2022-11-30:
Ok, the path that you see is not incorrect, that's where the program was compiled and that info is embedded as metadata into the assembly to make it easier to debug any problem.

And yes, the problem is your machine, the path where you have stored the application has restricted permissions and the app cannot read files from it. If you move it to another folder that does not have restricted permissions it will work.

Cheers.

closed 2022-11-30 by thisisgregm