gusmanb/logicanalyzer #35: LogicAnalyzer.exe does not show any channels

chriseast, 2022-12-07:
Hi, 

I downloaded and started the LogicAnalyzer.exe in Win10 and I'm able to connect to the Pico. The communication seems to work, since the Pico starts to blink when I hit accept in the capture settings and stops blinking when I hit abort.

Unfortunately the channels pane remains black all the time and I don't see any channels or captures.

What am I doing wrong?

gusmanb, 2022-12-07:
Hi.

Have you triggered any capture? The pico will stop blinking automatically once the trigger is detected, if it keeps blinking it means that it has not detected the trigger condition and you will see nothing, the device does not have any "continuous capture" mode as the USB would be too slow to transfer such ammount of data at the required speed.

Cheers.

gusmanb, 2023-01-13:
No more info has been provided in more than a month so this is being closed.

closed 2023-01-13 by gusmanb