gusmanb/logicanalyzer #37: osx-arm

tcurdt, 2023-01-13:
The releases only list `osx-x86` but theses days `osx-arm` is really needed as well.
Would be nice to also have that option.

gusmanb, 2023-01-13:
This is developed in .net 6.0 and it does not support osx-arm, only osx-x86, and as far as I know (I don't use MacOs) any x86 application should be supported in arm machines using Rosetta 2.

Have you tried this using Rosetta 2 and found any problem? If that's the case then please open a bug and I will check for it.

Cheers.

tcurdt, 2023-01-14:
I didn't try it for logicanalyzer.
Rosetta 2 should hopefully be a somewhat acceptable fallback - for now.
But it really isn't a long term solution.

Looking at https://dotnet.microsoft.com/en-us/download/dotnet/6.0 it seems like there is arm support for dotnet.
Hence I was hoping for a fully native release.

gusmanb, 2023-01-14:
Well, this is a happy coincidence, I just updated VS2022 from version 17.3 to 17.4.4 to add support for the new .net 7.0 SDK and now it lets me to build packages targeting osx-arm64, in fact I have now a ton more of targets like linux-arm64 (it had only arm in the previous version), windows-arm (besides windows-arm64) and so on, it seems that the framework indeed do support it but was Visual Studio the one that was not letting me to target them, not sure if it was my installation that was screwed or that the target has been added recently. 

I will create the packages with the next release.

Cheers.

gusmanb, 2023-01-29:
New release launched with many new targets including MacOSX-ARM64

closed 2023-01-29 by gusmanb

tcurdt, 2023-01-29:
I just downloaded the zip - but while `QLCapture` is recognized as executable

```
$ file CLCapture       
CLCapture: Mach-O 64-bit executable arm64
```

it does not start at all.

gusmanb, 2023-01-29:
Need more info as I don't have any mac around to test it...
Does it provide any error message? Have you sent any parameter? Have you tried to execute it as privileged user?

tcurdt, 2023-01-29:
TBH there isn't much to share.

```
$ ./CLCapture 
[1]    62402 killed     ./CLCapture
```

```
$ otool -L ./CLCapture 
./CLCapture:
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
```

Executing as privileged user "isn't really a thing" on macOS.
But starting it with "sudo" gives the same result.

This is on 12.6.3 on arm.

gusmanb, 2023-01-29:
"sudo" is "superuser do", what means to execute the program as a user with privileged permissions... that's a *nix thing common to all derivates :)

From the message looks like something killed the app, not an app crash. Have you tried the GUI app? This may give a bit more info, maybe this is a problem with .net and arm64, I have never used it so checking if the GUI app causes the same problem can at least point to a direction.

Cheers.

tcurdt, 2023-01-29:
> that's a *nix thing common to all derivates :)

That's why I did it ;)
But it's not something you would commonly do on macOS.

> From the message looks like something killed the app, not an app crash

Maybe it could be signing related. But that would usually pop up a security dialog.
It's a bit weird. I have never seen such behaviour before.

> From the message looks like something killed the app, not an app crash. Have you tried the GUI app?

What is "the GUI app"? I can't see anything in the dir that looks like a candidate.

```
CLCapture:                                             Mach-O 64-bit executable arm64
CLCapture.deps.json:                                   JSON data
CLCapture.dll:                                         PE32+ executable (console) Aarch64 Mono/.Net assembly, for MS Windows
CLCapture.pdb:                                         Microsoft Roslyn C# debugging symbols version 1.0
CLCapture.runtimeconfig.json:                          JSON data
CommandLine.dll:                                       PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
Microsoft.CSharp.dll:                                  PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
Microsoft.VisualBasic.Core.dll:                        PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
Microsoft.VisualBasic.dll:                             PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
Microsoft.Win32.Primitives.dll:                        PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
Microsoft.Win32.Registry.dll:                          PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
SharedDriver.dll:                                      PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
SharedDriver.pdb:                                      Microsoft Roslyn C# debugging symbols version 1.0
System.AppContext.dll:                                 PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Buffers.dll:                                    PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Collections.Concurrent.dll:                     PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Collections.Immutable.dll:                      PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Collections.NonGeneric.dll:                     PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Collections.Specialized.dll:                    PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Collections.dll:                                PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.ComponentModel.Annotations.dll:                 PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.ComponentModel.DataAnnotations.dll:             PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.ComponentModel.EventBasedAsync.dll:             PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.ComponentModel.Primitives.dll:                  PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.ComponentModel.TypeConverter.dll:               PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.ComponentModel.dll:                             PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Configuration.dll:                              PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Console.dll:                                    PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Core.dll:                                       PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Data.Common.dll:                                PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Data.DataSetExtensions.dll:                     PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Data.dll:                                       PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Diagnostics.Contracts.dll:                      PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Diagnostics.Debug.dll:                          PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
System.Diagnostics.DiagnosticSource.dll:               PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Diagnostics.FileVersionInfo.dll:                PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Diagnostics.Process.dll:                        PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Diagnostics.StackTrace.dll:                     PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Diagnostics.TextWriterTraceListener.dll:        PE32+ executable (DLL) (console) Mono/.Net assembly, for MS Windows
System.Diagnostics.Tools.dll:                          PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
[... cut ...]
libhostpolicy.dylib:                                   Mach-O 64-bit dynamically linked shared library arm64
libmscordaccore.dylib:                                 Mach-O 64-bit dynamically linked shared library arm64
libmscordbi.dylib:                                     Mach-O 64-bit dynamically linked shared library arm64
mscorlib.dll:                                          PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
netstandard.dll:                                       PE32 executable (DLL) (console) Intel 80386 Mono/.Net assembly, for MS Windows
```

gusmanb, 2023-01-29:
The CLCapture app is a command-line application only, it is for the cases that you may want to use the logic analyzer exclusively with sigrok or you don't have a graphical OS like a linux without any desktop manager, but the real application is the one called "LogicAnalyzer", it is a full fledged GUI for the device and a lot more versatile than the CL.

It is all on the releases section but here is the link to the osx-arm64 one: https://github.com/gusmanb/logicanalyzer/releases/download/V2.1.1.0/LogicAnalyzer-2.1.1.0-osx-arm64.zip

gusmanb, 2023-01-29:
Ok, I have found info about the problem, it is in fact a problem with code sign: https://github.com/dotnet/runtime/issues/79267

It has been a long time since I developed anything on mac but if I recall it right you need to have a developer account with Apple to do codesign, and I'm not going to pay 100€ just because apple wants it :P

You can try the steps proposed in the solution but if you don't have a developer license I doubt you will be able to sign the application:

# remove apple quarantine attribute
$ xattr -d com.apple.quarantine CLCapture

# set executable bit
$ chmod +x ./CLCapture

# code sign it
$ codesign --force --deep -s - ./CLCapture

Cheers.

tcurdt, 2023-01-29:
OK. I didn't realize there are two difference artifact downloads. My mistake!

Seems like all `dylib`s have to be removed from quarantine. 
Not ideal - but I was then able to start the application. 🎉

IIRC it used to be that an Apple account for signing was free - you could just not submit to the AppStore then.
But that might have changed.

Anyway - at least now it's clear how to get it running :)
Thanks for the help.

gusmanb, 2023-01-29:
There might be an intermediate solution, but I need to learn how to use Github Actions, it provides a build environment and it supports macOS as runner, it will sign the application with a self-signed certificate, it will not run at a first but at least it will show the warning about unknown developer and then from the finder you can open it with the context menu.

When I get some time and feel better I will give it a try.

Cheers!