gusmanb/logicanalyzer #40: Playback feature

ra1nb0w, 2023-01-25:
Hi,
can be nice to have a playback feature.
After you recorded the signal you can play it back to simulate the original device.
Do you think it is feasible and interesting?
thank you very much for the project!

gusmanb, 2023-01-25:
Hi!

Feasible is, while the device has the buffer in memory to replay it should be not really hard as it would be a very simple PIO program chained to a DMA that spits exactly what has captured, the problem here is the hardware, the shifters are unidirectional and by a good reason, I have used the bidirectional ones and they're a pain in the ass, the auto sensing feature works like crap and if by any chance both sides are at High-Z state they start sending voltage on both sides.

I could use bidirectional non-auto shifters, but that means a total rework of the device as there is no room for more signals (I would need to set the direction of the device depending on which function is used and that means at least one more signal to control the direction) in the connector and I would need to completelly redo the shifter board and main board to accomodate these changes.

I could add the replay feature but only for 3v3 and if I do that I'm sure that the people that uses the device with the shifter board would be pissed off...

When I finish the wireless support (it will take a bit longer than expected, I had a heart attack some weeks ago and right now the development is in pause) I will think about creating a totally new version of the device, as something that I would like to add is support for analog signals, basically convert the device into an oscilloscope, and that will need a total revamp of the device so it might be possible to add this feature.

Cheers!

ra1nb0w, 2023-01-27:
Thank you very much for the clarification.
In fact, having a feature that works only for one situation is not the best user experience.
If you rebuild the level shifter, for example adding the BNC connector for the oscilloscope functionality, I will suggest you to use the following components for the port:

- SN74LVC16T245
- SP3003 TVS diode array

Specially the second one to protect the device.

Anyway, thank you very much for the project!