gusmanb/logicanalyzer #45: Crash when exporting data, Windows

kuba2k2, 2023-02-13:
[image]

The program crashes whenever I try to export the data to a .lac or .csv file. I tested that on v2.0.0.0, v2.1.0.1 and v3.5.0.1 (latest).

Capturing data works fine, but I can't export it for further analysis in any way. How can I debug that crash?

gusmanb, 2023-02-14:
If you want to debug it you need to download the source code and use Visual Studio.
Maybe you can get info from the EventViewer, in the Application section it may have registered the crash.

When does it crash? Before or after selecting the filename to save/export?

Also, have you tried to execute it as administrator?

Cheers.

kuba2k2, 2023-02-14:
I'm afraid I don't currently meet the enormous disk space requirements of Visual Studio...

It crashes after selecting the filename and clicking OK, just in the moment when it should write the file. The file doesn't get created, even an empty one.

I have tried running both with admin rights and without.

Weirdly, it works just fine with the CLCapture program - the file gets created with all the data in it. I tried loading up the file in the GUI, and the same crash occurred (I know it expects a .lac and not .csv, so maybe that's the crash source in this case).

gusmanb, 2023-02-14:
Ok, let's see if we can find what's happening.

Take [this build](https://mega.nz/file/LSAX1KLa#qXuaeXXhbn11_Qdq2hKewNCCrEiCb__i8eYUQa4SMvw), instead of crashing it will show you the crash info before closing whenever you try to open/save/export a file. Test the three operations (opening a file that is not a .lac one is going to crash for sure as it is not in the correct format) and take a screenshot of each message, it will show exactly where and why the crash happens.

Also [here](https://github.com/gusmanb/logicanalyzer/files/10730360/captura.format.500Khz.zip) is a .lac file so you can test it properly.

BTW, this is built using my development branch and it contains something that I'm working on, so don't use the "New" menu under "File" or you will see strange things ;)

Cheers.

kuba2k2, 2023-02-14:
Same exception for all file operations:
[image]

I looked into Avalonia's source code of that file, and found that the exception is probably thrown when retrieving the chosen file's path. As I was using my D:\ drive for opening/saving the .lac files, I tried opening it from C:\ root directory instead, and... it worked.

For easy access to my dev directories, I have them set up as Windows Explorer "Libraries". Turns out that picking the file from the "library" crashes the program:
[image]

Picking the exact same file by manually navigating through PC->D:\->Dev->.... works just fine:
[image]

I'm not sure what's the cause of that behavior; I've written a few Windows programs that used file pickers before (in Python ~~and in Delphi~~) and picking files via a "library" simply returned the physical path of the file.

So, anyway, consider the issue half-solved. It's possible that on Windows 10 going the "libraries" way works and that it's a problem with Windows 7 (but I didn't test it so it might be broken as well). Thanks for the quick response and helping to solve the problem! :grinning:

gusmanb, 2023-02-14:
Excellent :)

I have tested it with a library in W10 and it works, so something must be different in the W7 libraries implementation. If I get enough time I will open a bug in AvaloniaUI.

Cheers!

closed 2023-02-14 by gusmanb