gusmanb/logicanalyzer #46: Serial port hangs if sampling interrupted

DevBruker, 2023-02-28:
Hi!
The serial port hangs if ClCapture is killed by Ctrl-C while the logic analyzer is sampling. ClCapture aborts, but ClCapture (and LogicAnalyzer) can not open the port unless the Pico is powered off and on. Same behaviour in version 3-5 as well.

This is on Ubuntu 22.04.

Using "Abort" in LogicAnalyzer works without issues, but gives the following message in the console:

The port is closed. -    at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count)
   at System.IO.Stream.Read(Span`1 buffer)
   at System.IO.BinaryReader.InternalRead(Int32 numBytes)
   at System.IO.BinaryReader.ReadUInt32()
   at SharedDriver.LogicAnalyzerDriver.ReadCapture(Int32 Samples, Byte Mode) in C:\Users\geniw\source\repos\LogicAnalyzer\SharedDriver\LogicAnalyzerDriver.cs:line 314

Thanks for a new version!

Regards

DevBruker, 2023-02-28:
Hi,
The new version behaves the same:
Executes in directory ...../logicanalyzer/release/25-feb-2023/fix/linux-x64 
./CLCapture capture /dev/ttyACM1 1000000 1,2,3,4,5,6,7,8 20 20000 TriggerType:Edge,Channel:1,Value:0 ../test.csv
Opening logic analyzer in /dev/ttyACM1...
Conneced to device LOGIC_ANALYZER_V4_0 in port/address /dev/ttyACM1
Starting edge triggered capture...
Capture running...
^C
./CLCapture capture /dev/ttyACM1 1000000 1,2,3,4,5,6,7,8 20 20000 TriggerType:Edge,Channel:1,Value:0 ../test.csv
Opening logic analyzer in /dev/ttyACM1...
Error detecting Logic Analyzer in port/address /dev/ttyACM1

Regards

gusmanb, 2023-03-01:
Ok, let's see if this one works for you, I have tested it under a vm with Debian and it worked. It seems that under Linux the serial class behaves differently than in Windows and it doesn't flushes correctly the data and leaves the thread hung. I have added some delays and changed a bit how the cancel works in the driver, it will take a couple of seconds between you pressing Ctrl+c and the actual abort and after that it will show a message telling that the capture has been aborted.

https://mega.nz/file/vHIBzDZT#fXRTpeDyTOXGC6yHfadQARfAAS3wo__IIncZb0FNf0Y

gusmanb, 2023-03-01:
Ok, let's see if this one works for you, I have tested it under a vm with Debian and it worked. It seems that under Linux the serial class behaves differently than in Windows and it doesn't flushes correctly the data and leaves the thread hung. I have added some delays and changed a bit how the cancel works in the driver, it will take a couple of seconds between you pressing Ctrl+c and the actual abort and after that it will show a message telling that the capture has been aborted.

https://mega.nz/file/ae4mXICa#PhkkylmZlIFstFZ6VqUdRZePu3mjS1k4t6IL-7y6vPY

DevBruker, 2023-03-01:
Thanks, works as you said:

/CLCapture capture /dev/ttyACM1 1000000 1,2,3,4,5,6,7,8 20 131051 TriggerType:Edge,Channel:1,Value:0 ../test.csv
Opening logic analyzer in /dev/ttyACM1...
Conneced to device LOGIC_ANALYZER_V4_0 in port/address /dev/ttyACM1
Starting edge triggered capture...
Capture running...
^CThe port is closed. -    at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count)
   at System.IO.Stream.Read(Span`1 buffer)
   at System.IO.Stream.ReadAtLeastCore(Span`1 buffer, Int32 minimumBytes, Boolean throwOnEndOfStream)
   at System.IO.BinaryReader.InternalRead(Int32 numBytes)
   at System.IO.BinaryReader.ReadUInt32()
   at SharedDriver.LogicAnalyzerDriver.ReadCapture(Int32 Samples, Byte Mode) in C:\Users\geniw\source\repos\LogicAnalyzer\SharedDriver\LogicAnalyzerDriver.cs:line 314
Capture aborted.

Regards

gusmanb, 2023-03-01:
Excellent.

I have uploaded the code to the repo and when I get a bit of time will do a small release with the patch.

Cheers!

closed 2023-03-01 by gusmanb