gusmanb/logicanalyzer #52: Window Height Not Adjustable

ramblejeff, 2023-03-05:
My maximum screen height is 768. The window height, defaults to 800 and can not be adjusted. Altering the "Height" in AppConfig.json:
 
"WindowSettings": {
    "LogicAnalyzer.MainWindow": {
      "Width": 1024.0,
      "Height": 800.0,

 has no effect, when the program is closed the height is reset to 800. Is there somewhere where I can tweak this?

gusmanb, 2023-03-05:
The window has a fixed minimum bounds, it's designed for modern screens that usually are at least 1920x1080 pixels...

Which OS are you using?

ramblejeff, 2023-03-06:
Kubuntu on my workshop Laptop.  The display is 1366 x 768.  I know its an older machine that is what a lot of us have.

Otherwise the program runs fine and the wiki is helpful.

gusmanb, 2023-03-06:
Ok, let's see if this works ok.
https://mega.nz/file/XaBhlDAR#8aYUJw2KYtH035y62kdV2u2sOJl05-kdGb_6VfDbuWk

I have reduced the minimum window size to 800x600, the view area is going to be small but it will fit in your screen.

Cheers!

ramblejeff, 2023-03-06:
Thanks,

      Works well.
Jeff.

https://mega.nz/file/XaBhlDAR#8aYUJw2KYtH035y62kdV2u2sOJl05-kdGb_6VfDbuWk

closed 2023-03-06 by ramblejeff