gusmanb/logicanalyzer #57: Wish

Kabron287, 2023-05-07:
It'll be very useful to use ADC to measure reference voltage of the level shifter 
and use PWM or external DAC to setup reference voltage directly from the SW.

gusmanb, 2023-05-07:
The ADC will be fried if the external voltage is higher than 3.3v, and using PWM to power the shifters can cause a lot of troubles. Also, there is no free pin left...

closed 2023-05-07 by gusmanb

Kabron287, 2023-05-07:
>> Dividers exists.

gusmanb, 2023-05-07:
> > The ADC will be fried if the external voltage is higher than 3.3v
> 
> Dividers exists.

Cool! PR's also do exist, this is an open-source project so feel free to design it, implement it, test it, all in a way that doesn't disturbs any of the other functionalities and then do a PR wit the changes, if all works I will incorporate it to the design.