gusmanb/logicanalyzer #60: missing executable permission on Linux-x64 binaries

bablokb, 2023-05-27:
Hi,

thanks for providing this repo. I am currently trying to install it on my Linux system and stumbled over one problem:
the main executables in the zipped releases are missing the executable-bit.

Doing a `chmod +x linux-64/LogicAnalyzer` solves the problem (same for the cli-interface). After that, it starts :-) 

Bernhard

gusmanb, 2023-05-27:
Hi.

Yes, it is known, I compile the programs under Windows so I can't set the permissions for Linux, it is explained in the README: https://github.com/gusmanb/logicanalyzer#update-13072022

Cheers.

closed 2023-05-27 by gusmanb

bablokb, 2023-05-27:
Oh yes. But this is a very long readme with a lot of updates. Most users won't really be interested in history after all and won't pay attention to all the details written there.

So I suggest that you add this important information  to the wiki. Like other stuff too, e.g. the basic setup you also describe in the readme way down almost at the beginning (I was also looking for that in the wiki, did not find it but managed to run the basic setup without this information).