gusmanb/logicanalyzer #63: Some LogicAnalyser problems/suggestions

bablokb, 2023-06-03:
The LogicAnalyser does not keep the configured state:

on recapture, it looses the setting "samples in screen". What is more annoying is that it also looses the information about the protocol analyzer. Having to tell the program for every capture that I want to do an I2C-analysis and having to redefine the SDA/SCL channels is cumbersome.

It would be great if this would be kept as part of the configuration. Some more things on the whishlist:
  - accept a file with a configuration on the commandline. 
  - if I add SDA/SCL to the channel description, take this as a hint and configure the I2C protocol analyzer (same for other protocols)
  - add a sort of clickable capture map where I can see where something is happening. Of course, scrolling to the right works fine, but this would help navigating to the correct places. Something like a keyboard shortcut "skip right to the next trigger" would also be great.
  - AppConfig.json and cpSettingsSerial.json are always written in the current work-directory. I have these files now all over the  place. They should ideally go to $HOME/.config/LogicAnalyzer. Or at least always to the same directory (don't know if multiplatform binaries support this in a generic way, but Windows has it's standards, Linux and I suppose Apple as well, so it could be there is some sort of abstraction available).

Another note: the visual design does not seem to be very ergonomic (very low contrasts of controls). In fact in my first experiments I missed the scrollbar on the bottom, because I  could not identify it. But maybe this is Linux specific.

gusmanb, 2023-06-16:
Hi.

Thanks for the suggestions.

I have implemented some changes that may mitigate your problems.

First, the samples in screen will now be preserved whenever you use the "Repeat last capture", in case of a new capture or restarting the program it will still reset the samples in screen to the calculated value based in the samples captured.

Next, I have also added a new "Repeat last analysis" menu entry, so once you have analyzed for the first time a capture you will be able to repeat the same analysis with a single click. This is also only stored per session and will not be persisted. Beware that the analyzers are based on the channel index, they don't use at all the channel names, those are purely aesthetic for the user to distinguish the signals, so if you capture the channels in a different position the analysis will still use the old indexes even if you change the channel names.

Also, the config files should now be stored in the application folder ignoring the current path.

Finally I have made the horizontal scrollbar always visible to make it easier to be seen.

Here is the new version, could you test it in Linux and confirm that it works correctly? Right now I don't have any Linux machine around and I have tested it only in Windows.

https://mega.nz/file/DHpXyTZb#y1SybBT0ypeOQboa94v-n3ugwQzxL5LGhxQQPEuOwQs

You will need to copy the "analyzers" folder from the distribution package.

Cheers.

bablokb, 2023-06-17:
Thanks! I downloaded it already and will test it.

BTW: I just finished a review for this tool for the German magazine "Raspberry-Pi-Geek". It should be in print in about two months, unless they move it to a later issue.

bablokb, 2023-06-17:
I tested the new version:

- samples: works
- repeat last analysis: works
- horizontal scrollbar: much  better
- configuration files: fails

Regarding the configuration files: AppConfig.json is still written to the current directory. And the program tries to write cpSettingsSerial.json to the application folder. Which is owned by root. So this fails and the program dumps. Separation of programs and (user) data is not specific to Linux, you would also have the same problem running Windows if the software is cleanly installed in "C:\Program Files".

One option would be to provide a runtime option (e.g. -c dir) for the configuration files. I can imagine that it makes sense to store multiple versions for different settings. If -c is not specified, default to $Home/.config/LogicAnalyzer or something below %appdata% on Windows (no idea about  Apple, but they will also have a default dir). 

Thanks for your work on this, the first two items already save a lot of clicking and are very valuable!

gusmanb, 2023-06-17:
Hi.

Thanks for testing.

I was trying to preserve all the files in the same folder as the idea was to have a portable app (there is no installer and the application is a self-contained one, it does not need you to install nothing, not even .net framework), that's why I tried to write to the application folder, in this way you could have the app in (for example) an USB stick and use it in different machines. Also, I made a mistake saving the appconfig and it was not trying to store it in the new folder.

In any case, I have changed it, now it uses the appData folder to store the config files (in linux it will store it in $home/.config/LogicAnalyzer).

Here is the link to the new version.

https://mega.nz/file/XbxTVBqR#C3rflcrBzIPV_klC7UkzLyFc05zGXMEMcSDJxD93L6A

Cheers!

bablokb, 2023-06-17:
Hi, thanks. I will test this on Monday.

bablokb, 2023-06-19:
Perfect! Thanks!

gusmanb, 2023-06-19:
Excellent, I will release the changes this weekend in a new version, also I will port the config changes to the CLI app.

Cheers!

gusmanb, 2023-07-25:
@bablokb FYI, I have added shortcuts to navigate the samples with keyboard and also a preview of the full captured data that pops up when you hover the scroll bar to ease the navigation. More info [here](https://github.com/gusmanb/logicanalyzer/wiki/06---The-LogicAnalyzer-program#navigating-on-the-capture-viewer)

Cheers!

bablokb, 2023-07-27:
Great addition, this really helps to gain an initial overview.