gusmanb/logicanalyzer #65: Getting a crash in osx-64

tony1tf, 2023-06-25:
Hi
Thanks for this wonderful software. I've gone through the steps in previous discussions to run the Logic analyser s/w for osx-64, and have got it running on a Macbook with M2 silicon. There doesn't seem to be an option to run it without a trigger and I havn't got anything connected to the pico yet. Anyway, I can discover the pico and ask to collect some samples at 4Msps. However, after waiting for some seconds , it crashes saying Logic Analyser quit unexpectedly with the following first few lines of the dump. (there's many more!). Any thoughts?
Tony

Anonymous UUID:        89536F21-3E73-9092-67DE-18D78A68982B

Sleep/Wake UUID:       BA24C633-4ED1-4FB3-A18A-A1EBCB8B9EAF

Time Awake Since Boot: 530000 seconds
Time Since Wake:       3088 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x197234724 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19726bc28 pthread_kill + 288
2   libsystem_c.dylib             	       0x197179ae8 abort + 180
3   libcoreclr.dylib              	       0x1049bdb34 PROCAbort + 60
4   libcoreclr.dylib              	       0x1049bda44 PROCEndProcess(void*, unsigned int, int) + 396
5   libcoreclr.dylib              	       0x104be921c UnwindManagedExceptionPass1(PAL_SEHException&, _CONTEXT*) + 968
6   libcoreclr.dylib              	       0x104be9264 DispatchManagedException(PAL_SEHException&, bool) + 68
7   libcoreclr.dylib              	       0x104b4ab5c IL_Throw(Object*) + 528
8   ???                           	       0x2830265c4 ???
9   ???                           	       0x282970098 ???
10  ???                           	       0x28296fa74 ???
11  ???                           	       0x282874064 ???
12  ???                           	       0x280db677c ???
13  ???                           	       0x280d85e3c ???
14  libcoreclr.dylib              	       0x104c72408 CallDescrWorkerInternal + 132
15  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
16  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
17  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
18  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
19  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
20  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852
21  libcoreclr.dylib              	       0x104ae3a98 MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 852

gusmanb, 2023-06-26:
Hi.

I would need a full log to check the problem, in this log the only thing that I can see is that something failed inside a thread, so I assume the problem is happening in the capture driver thread, but without knowing which function has crashed is very difficult to see what is happening.

Also, are you using sudo to execute the application? it may be a problem with privileges so it would be worth it to test it.

Finally, if you don't have anything connected to the analyzer but want to try if the capture works you can configure a trigger for negative edge, as nothing is connected to the inputs it will trigger immediatelly, all the channels will capture zeros but it will be enough for testing purposes.

Cheers.

tony1tf, 2023-06-26:
ps - I have tried enabling wifi on a PicoW, and the application shows the correct type, but the wifi tab doesn't open. Screenshot attached.
[image]

Tony

gusmanb, 2023-06-26:
Hi.

Ok, now I see your problem, you're using a very old software package, that version is not compatible with the WiFi version (the menu is missing) and the device is not answering to the capture request as it uses a previous protocol which causes the crash.

Download the latest package from the [V4.0.0.0](https://github.com/gusmanb/logicanalyzer/releases/tag/v4.0.0.0) and it will work.

Cheers.

tony1tf, 2023-06-26:
I tried this, but it didn't make any difference.
codesign --force --deep -s - ./libhostfxr.dylib

gusmanb, 2023-06-26:
Hi.

Welcome to the "wonders" of the new codesign enforcement for ARM Mac's... That's because the application is not signed (I'm not going to pay Apple a developer license for a free program :P) and as Apple doesn't gets money for the usage of the app they block it :/

There are workarounds, you can right-click the app while holding Control and then use "Open", another option is to manually remove the quarantine attribute from the files.

Here you can find a tutorial on how to do it: 
https://www.easeus.com/computer-instruction/apple-cannot-check-it-for-malicious-software.html

If you remove the quarantine attribute you must remove it from all the libs and the executable.

Another option is to use the osx-64 version instead of the ARM one, that one will be executed using Rosetta and does not enforce these requirements.

Cheers.

tony1tf, 2023-06-26:
Hi Agustin

Although these measures let me run 'LogicAnalyzer' ,  the Macbook is always failing to run libhostfxr.dylib
I don't seem to be able to remove quarantine from the other files.
The errors were suggesting I should download dotnet:

The library libhostfxr.dylib was found, but loading it from /Users/Tony/RPi_Pico/Another_Logic_Analyser/osx-arm64/libhostfxr.dylib failed
  - Installing .NET prerequisites might help resolve this problem.
     https://go.microsoft.com/fwlink/?linkid=2063366

I've tried the osx-64 version and have the same problem - I can open LogicAnalyzer, but no farther.

I'm sorry that I am out of my depth here.

Tony

tony1tf, 2023-06-26:
I have finally sorted it (I think) - global setting for the whole directory wasn't working, so I removed quarantine from all the dylib files manually, then had to do the same for the dll's. I was able to do this with:
xattr -d com.apple.quarantine *.dll
I have been able to set an IP address in the wifi enabled one and capture all zeroes - thank you for all your help.
Tony

tony1tf, 2023-06-26:
The command need the apple quarantine name in it - this worked:
sudo xattr -d -r com.apple.quarantine /Users/Tony/RPi_Pico/Another_Logic_Analyser/osx-arm64
So now the full arm64 version is working OK.
Thank you
Tony

gusmanb, 2023-06-26:
Thanks for the confirmation.

I'm going to add this info to the wiki so it's easier for others that want to use the program in a Mac.

Cheers!