gusmanb/logicanalyzer #66: linux x86 runtime crash

jixishi, 2023-06-30:
System Information:
[image]
```
dotnet --version
7.0.107
```

Error message:
```./LogicAnalyzer
Unhandled exception. System.InvalidOperationException: Default font family name can't be null or empty.
   at Avalonia.Media.FontManager..ctor(IFontManagerImpl platformImpl) in /_/src/Avalonia.Visuals/Media/FontManager.cs:line 33
   at Avalonia.Media.FontManager.get_Current() in /_/src/Avalonia.Visuals/Media/FontManager.cs:line 53
   at Avalonia.Media.Typeface.get_GlyphTypeface() in /_/src/Avalonia.Visuals/Media/Typeface.cs:line 69
   at Avalonia.Media.TextFormatting.TextCharacters.TryGetRunProperties(ReadOnlySlice`1 text, Typeface typeface, Typeface defaultTypeface, Int32& count) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 132
   at Avalonia.Media.TextFormatting.TextCharacters.CreateShapeableRun(ReadOnlySlice`1 text, TextRunProperties defaultProperties) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 62
   at Avalonia.Media.TextFormatting.TextCharacters.GetShapeableCharacters() in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 40
   at Avalonia.Media.TextFormatting.TextFormatterImpl.FetchTextRuns(ITextSource textSource, Int32 firstTextSourceIndex, TextLineBreak previousLineBreak, TextLineBreak& nextLineBreak) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextFormatterImpl.cs:line 286
   at Avalonia.Media.TextFormatting.TextFormatterImpl.FormatLine(ITextSource textSource, Int32 firstTextSourceIndex, Double paragraphWidth, TextParagraphProperties paragraphProperties, TextLineBreak previousLineBreak) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextFormatterImpl.cs:line 15
   at Avalonia.Media.TextFormatting.TextLayout.UpdateLayout() in /_/src/Avalonia.Visuals/Media/TextFormatting/TextLayout.cs:line 225
   at Avalonia.Media.TextFormatting.TextLayout..ctor(String text, Typeface typeface, Double fontSize, IBrush foreground, TextAlignment textAlignment, TextWrapping textWrapping, TextTrimming textTrimming, TextDecorationCollection textDecorations, Double maxWidth, Double maxHeight, Double lineHeight, Int32 maxLines, IReadOnlyList`1 textStyleOverrides) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextLayout.cs:line 71
   at Avalonia.Controls.TextBlock.CreateTextLayout(Size constraint, String text) in /_/src/Avalonia.Controls/TextBlock.cs:line 475
   at Avalonia.Controls.Primitives.AccessText.CreateTextLayout(Size constraint, String text) in /_/src/Avalonia.Controls/Primitives/AccessText.cs:line 156
   at Avalonia.Controls.TextBlock.get_TextLayout() in /_/src/Avalonia.Controls/TextBlock.cs:line 166
   at Avalonia.Controls.TextBlock.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/TextBlock.cs:line 521
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Presenters.ContentPresenter.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ContentPresenter.cs:line 366
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Border.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Border.cs:line 187
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Controls.StackPanel.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/StackPanel.cs:line 207
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Controls.Presenters.ItemVirtualizer.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ItemVirtualizer.cs:line 221
   at Avalonia.Controls.Presenters.ItemsPresenter.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ItemsPresenter.cs:line 139
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Border.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Border.cs:line 187
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Controls.DockPanel.MeasureOverride(Size constraint) in /_/src/Avalonia.Controls/DockPanel.cs:line 112
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Presenters.ContentPresenter.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ContentPresenter.cs:line 366
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Controls.Decorator.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Decorator.cs:line 54
   at Avalonia.Controls.Primitives.VisualLayerManager.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Primitives/VisualLayerManager.cs:line 133
[... cut ...]
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at LogicAnalyzer.Program.Main(String[] args) in C:\Users\geniw\source\repos\LogicAnalyzer\LogicAnalyzer\Program.cs:line 14
zsh: IOT instruction (core dumped)  ./LogicAnalyzer
```

gusmanb, 2023-06-30:
This is a known problem of Avalonia with some OS'es/lang configs.

https://github.com/AvaloniaUI/Avalonia/issues/4427

As you use a derivate of Arch, according to the post installing ttf-dejavu font should solve the problem.

closed 2023-07-02 by gusmanb

taotieren, 2023-08-28:
I've added the [logicanalyzer-git](https://aur.archlinux.org/packages/logicanalyzer-git) package to the AUR so you can test it out.

Ragebone, 2024-01-02:
@taotieren 
On a fresh install, the application crashes because access is denied.
Manjaro and stuff, should be an easy fix and this is probably not the right place but better this then me plowing on and forgetting about it again. 
```
--- Begin breadcrumb write
Directory core breadcrumbs [] was not specified or found
Fallback directory core breadcrumbs at [opt/corebreadcrumbs] was not found
Breadcrumb store was not obtained... skipping write.
Unhandled exception. System.UnauthorizedAccessException: Access to the path '/usr/share/logicanalyzer/cpSettingsSerial.json' is denied.
 ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirError)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, UnixFileMode openPermissions, Int64& fileLength, UnixFileMode& filePermissions, Func`4 createOpenException)
   at System.IO.File.OpenHandle(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.File.WriteToFile(String path, FileMode mode, String contents, Encoding encoding)
   at LogicAnalyzer.Dialogs.CaptureDialog.btnAccept_Click(Object sender, RoutedEventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 199
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 245
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at LogicAnalyzer.Program.Main(String[] args)
/usr/bin/logicanalyzer: line 4: 38399 Aborted                 (core dumped) LC_ALL=C ./LogicAnalyzer
```