gusmanb/logicanalyzer #70: Some more suggestions

bablokb, 2023-07-27:
Hi,

some more suggestions:

- add a help-menu with a link to the wiki and an "About-entry" with the current version-number/build number
- remember the IP-address of the last network-device
- add D/C channel to SPI analyzer

The first one is "nice to have", the second one is useful. I think most users will typically have one device only. The third one would help in analyzing the 4-wire protocol (most SPI stuff I am using has this data/command line).

(as you can see: users always want more features ;-)

gusmanb, 2023-07-28:
Ok, i will give it a try this weekend and implement it.

Cheers!

gusmanb, 2023-10-24:
Implemented and commited, will be available in the next release

closed 2023-10-24 by gusmanb