gusmanb/logicanalyzer #73: Cannot find my device

ericzhang-debug, 2023-08-15:
Hello,
My device is RPi-PICO which was bought in 2020. I burned the firmware with the version of 5.0.0(Firmware-LogicAnalyzer-5.0.0.0-PICO.uf2). But when i click the program in windows with the device connected to my pc and a terminal flashed and exited. I tried to exec it in terminal and it showed ERROR(S): No verb selected.

[image]

And the device in Device Manager shows RESET,which needs related drivers.
[image]

What can i do to solve it?
Thank you very much!

gusmanb, 2023-08-15:
Hi.

The device does not need any driver, it's detected as a standard USB serial device.

The CLCapture is the command line capture program, you must provide a full command to use it, as it states the first that you must provide is a verb, for example "capture", if you want to use the cl program I recommend you to read the [wiki entry](https://github.com/gusmanb/logicanalyzer/wiki/05---The-CLCapture-program) that explains its usage.

In any case I suspect you want to use the GUI program, that's the "LogicAnalyzer" package that you can find in the [releases](https://github.com/gusmanb/logicanalyzer/releases) page. Maybe you missed it as each release contains 20 packages and Github only shows the 10 first entries, to see all you need to click on "Show all 20 assets".

Cheers.

ericzhang-debug, 2023-08-15:
Well, I solve it .Thank you very much!

closed 2023-08-15 by ericzhang-debug