gusmanb/logicanalyzer #76: Failure to compile dotnet to a single file

taotieren, 2023-08-30:
I have packaged [logicanalyzer](https://aur.archlinux.org/packages/logicanalyzer-git) in AUR repository .
Now the compiled file is too fragmented, I tried to compile it into a single file and now the compilation reports an error. 

PKGBUILD

```bash
# Maintainer: taotieren <admin@taotieren.com>

pkgname=logicanalyzer-git
pkgver=5.0.0.0.r1.g5f95102
pkgrel=2
epoch=
pkgdesc="24 channel, 100Msps logic analyzer hardware and software "
arch=("x86_64" "aarch64")
url="https://github.com/gusmanb/logicanalyzer"
license=('GLP-3.0')
provides=(${pkgname%-git} clcapture)
conflicts=(${pkgname%-git} clcapture)
replaces=()
depends=(dotnet-runtime
    ttf-dejavu)
makedepends=(git
    dotnet-sdk)
backup=()
options=('!strip')
install=
source=("${pkgname-git}::git+${url}.git")
sha256sums=('SKIP')
noextract=()

pkgver(){
    cd "${srcdir}/${pkgname-git}"
    git describe --long --tags | sed 's/^V//g;s/\([^-]*-g\)/r\1/;s/-/./g'
}

build() {
    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer"

    if [ "$CARCH" == "aarch64" ]; then
    msg2 "build for arm64"
        dotnet publish  -r linux-arm64 -c Release -p:PublishSingleFile=true -p:PublishTrimmed=true --self-contained true -o artifact
#         dotnet publish  -r linux-arm64 -c Release -o artifact
    else
    msg2 "build for x64"
        dotnet publish  -r linux-x64 -c Release -p:PublishSingleFile=true -p:PublishTrimmed=true --self-contained true -o artifact
#         dotnet publish  -r linux-x64 -c Release -o artifact
    fi
}

package() {
    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/"

    install -Dm0755 CLCapture "${pkgdir}/usr/bin/clcapture"
#     install -Dm0644 CLCapture.dll "${pkgdir}/usr/share/${pkgname%-git}/CLCapture.dll"
    install -Dm0755 LogicAnalyzer "${pkgdir}/usr/bin/logicanalyzer"
#     install -Dm0644 LogicAnalyzer.dll "${pkgdir}/usr/share/${pkgname%-git}/LogicAnalyzer.dll"
#     install -Dm0644 LogicAnalyzer.runtimeconfig.json "${pkgdir}/usr/share/${pkgname%-git}/LogicAnalyzer.runtimeconfig.json"

#     install -dm0755 "${pkgdir}/usr/bin" \
#        install -dm0755 ${pkgdir}/usr/share/${pkgname%-git}
#
#     install -Dm0755 /dev/stdin  "${pkgdir}/usr/bin/clcapture" <<EOF
# export COREHOST_TRACE=1
#
# cd  /usr/share/logicanalyzer/
# LC_ALL=C ./CLCapture
# EOF
#
#     install -Dm0755 /dev/stdin  "${pkgdir}/usr/bin/${pkgname%-git}" <<EOF
# export COREHOST_TRACE=1
[... 70 lines cut ...]
/usr/share/dotnet/sdk/7.0.110/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.Publish.targets(94,5): error NETSDK1099: Publishing to a single-file is only supported for executable applications. [/home/taotieren/git_clone/aur/logicanalyzer-git/src/logicanalyzer-git/Software/LogicAnalyzer/I2CProtocolAnalyzer/I2CProtocolAnalyzer.csproj]
  SerialProtocolAnalyzer -> /home/taotieren/git_clone/aur/logicanalyzer-git/src/logicanalyzer-git/Software/LogicAnalyzer/SerialProtocolAnalyzer/bin/Release/net7.0/linux-x64/SerialProtocolAnalyzer.dll
/usr/share/dotnet/sdk/7.0.110/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.Publish.targets(94,5): error NETSDK1099: Publishing to a single-file is only supported for executable applications. [/home/taotieren/git_clone/aur/logicanalyzer-git/src/logicanalyzer-git/Software/LogicAnalyzer/SerialProtocolAnalyzer/SerialProtocolAnalyzer.csproj]
  SPIProtocolAnalyzer -> /home/taotieren/git_clone/aur/logicanalyzer-git/src/logicanalyzer-git/Software/LogicAnalyzer/SPIProtocolAnalyzer/bin/Release/net7.0/linux-x64/SPIProtocolAnalyzer.dll
/usr/share/dotnet/sdk/7.0.110/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.Publish.targets(94,5): error NETSDK1099: Publishing to a single-file is only supported for executable applications. [/home/taotieren/git_clone/aur/logicanalyzer-git/src/logicanalyzer-git/Software/LogicAnalyzer/SPIProtocolAnalyzer/SPIProtocolAnalyzer.csproj]
==> ERROR: A failure occurred in build().
    Aborting...

```

```bash
 exa -Tla src/logicanalyzer-git/Software/LogicAnalyzer/artifact/
drwxr-xr-x    - taotieren 30 Aug 10:10 src/logicanalyzer-git/Software/LogicAnalyzer/artifact
.rwxr-xr-x  14M taotieren 30 Aug 10:10 ├── CLCapture
.rw-r--r--  13k taotieren 30 Aug 10:10 ├── CLCapture.pdb
.rwxr--r-- 1.6M taotieren 12 Aug  2022 ├── libHarfBuzzSharp.so
.rwxr--r-- 683k taotieren 18 Feb  2022 ├── libonigwrap.so
.rwxr--r-- 9.2M taotieren 12 Aug  2022 ├── libSkiaSharp.so
.rwxr--r--  18k taotieren 23 Oct  2021 ├── libSystem.IO.Ports.Native.so
.rwxr-xr-x  51M taotieren 30 Aug 10:10 ├── LogicAnalyzer
.rw-r--r--  73k taotieren 30 Aug 10:10 ├── LogicAnalyzer.pdb
.rw-r--r--  17k taotieren 30 Aug 10:10 ├── SharedDriver.pdb
.rw-r--r--  16k taotieren 30 Aug 10:10 └── SignalDescriptionLanguage.pdb

```

gusmanb, 2023-08-30:
Hi.

The protocol analyzers can't be compiled into single file executables as they are standalone DLL's. These basically are plugins that are loaded by the LogicAnalyzer app on the fly using reflection, so they need to be built separatedly and placed in the folder "analyzers" on the same fodler the application resides in order for them to work.

Also, I would not recommend to use single file executables, these are problematic when reflection is used and I can't guarantee that Avalonia works with them.

Finally, you are mixing different applications into a single package.
CLCapture and LogicAnalyzer are different apps with different purposes, one (CLCapture) is intended for those people that doesn't wants to or cannot use the GUI app and will use a third-party software like PulseView to analyze the captures, the other (LogicAnalyzer) is the main GUI application and does not need at all CLCapture if is used.

Cheers.

taotieren, 2023-08-30:
Thank you. Could you add the ability to execute a single file to `CLCapture`? This would reduce the number of files to be installed and make it easier for users to use.

gusmanb, 2023-08-30:
Hi.

CLCapture can already be built as single file, you just need to issue the correct commands. Keep in mind that even if you build it as "single file" it will produce multiple ones as the native libraries will still be outside the packaged file.

To build the two programs with the correct structure you should update the build step like this (I'm posting only the x64 version, expand it as you wish to include other architectures, maybe use a var so you don't need to create twice the code):

```
build() {

    #first step, prepare the structure

    mkdir "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact"
    mkdir "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer"
    mkdir "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"
    mkdir "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/CLCapture"

    #second step, compile analyzers. These are pure IL assemblies, do not need to publish for a specified architecture
    
    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/SerialProtocolAnalyzer"
    dotnet publish  -c Release -o artifact
    cp ./artifact/SerialProtocolAnalyzer.dll "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/I2CProtocolAnalyzer"
    dotnet publish  -c Release -o artifact
    cp ./artifact/I2CProtocolAnalyzer.dll "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/SPIProtocolAnalyzer"
    dotnet publish  -c Release -o artifact
    cp ./artifact/SPIProtocolAnalyzer.dll "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    #third step, compile LogicAnalyzer

    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/LogicAnalyzer"
    dotnet publish  -r linux-x64 -c Release --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/LogicAnalyzer"

    #fourth step, compile CLCapture

    cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/CLCapture"
    dotnet publish  -r linux-x64 -c Release -p:PublishSingleFile=true -p:PublishTrimmed=true --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgname-git}/Software/LogicAnalyzer/artifact/CLCapture"

}
```

With this build you will end with a structure like this:

```
gusman@DESKTOP-C983CQ7:/mnt/c/Users/geniw/source/repos/LogicAnalyzer/artifact$ exa -Tla
drwxrwxrwx    - gusman 30 Aug 10:08 .
drwxrwxrwx    - gusman 30 Aug 10:24 ├── CLCapture
.rwxrwxrwx  14M gusman 30 Aug 10:24 │  ├── CLCapture
.rwxrwxrwx  18k gusman 23 Oct  2021 │  └── libSystem.IO.Ports.Native.so
drwxrwxrwx    - gusman 30 Aug 10:38 └── LogicAnalyzer
drwxrwxrwx    - gusman 30 Aug 10:11    ├── analyzers
.rwxrwxrwx 9.2k gusman 30 Aug 10:11    │  ├── I2CProtocolAnalyzer.dll
.rwxrwxrwx 9.2k gusman 30 Aug 10:10    │  ├── SerialProtocolAnalyzer.dll
.rwxrwxrwx  12k gusman 30 Aug 10:11    │  └── SPIProtocolAnalyzer.dll
.rwxrwxrwx  68k gusman 15 Aug  2022    ├── Avalonia.Animation.dll
.rwxrwxrwx 301k gusman 15 Aug  2022    ├── Avalonia.Base.dll
.rwxrwxrwx 962k gusman 15 Aug  2022    ├── Avalonia.Controls.dll
.rwxrwxrwx 218k gusman 15 Aug  2022    ├── Avalonia.DesignerSupport.dll
.rwxrwxrwx  12k gusman 15 Aug  2022    ├── Avalonia.Desktop.dll
.rwxrwxrwx  36k gusman 15 Aug  2022    ├── Avalonia.DesktopRuntime.dll
.rwxrwxrwx 252k gusman 15 Aug  2022    ├── Avalonia.Dialogs.dll
.rwxrwxrwx  10k gusman 15 Aug  2022    ├── Avalonia.dll
.rwxrwxrwx  94k gusman 15 Aug  2022    ├── Avalonia.FreeDesktop.dll
.rwxrwxrwx 119k gusman 15 Aug  2022    ├── Avalonia.Input.dll
.rwxrwxrwx  29k gusman 15 Aug  2022    ├── Avalonia.Interactivity.dll
.rwxrwxrwx  93k gusman 15 Aug  2022    ├── Avalonia.Layout.dll
.rwxrwxrwx  72k gusman 15 Aug  2022    ├── Avalonia.Markup.dll
.rwxrwxrwx  72k gusman 15 Aug  2022    ├── Avalonia.Markup.Xaml.dll
[... 236 lines cut ...]
   at System.IO.Path.Combine(String path1, String path2)
   at LogicAnalyzer.MainWindow.LoadAnalyzers()
   at LogicAnalyzer.MainWindow..ctor()
   at LogicAnalyzer.App.OnFrameworkInitializationCompleted()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 181
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Threading.Dispatcher.RunJobs() in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 73
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 199
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 245
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
[... cut ...]

- Single file apps are slower and consume more resources.

- You can remove the dotnet-runtime dependency, when you specify a self-contained application the app already contains all the required libraries and it does not depend on the dotnet runtime, that's why it weights more, but I always prefer it like this as it avoids compatibility problems.

- I had to remove the trim from LogicAnalyzer as Avalonia does not like it at all, when trimmed the app crashes when initializing Avalonia.

Cheers!

taotieren, 2023-08-30:
Here's the organized PKGBUILD file, thanks!

```bash
# Maintainer: taotieren <admin@taotieren.com>

pkgbase=logicanalyzer-git
pkgname=({logicanalyzer,clcapture}-git)
pkgver=5.0.0.0.r1.g5f95102
pkgrel=2
epoch=
pkgdesc="24 channel, 100Msps logic analyzer hardware and software "
arch=("x86_64" "aarch64")
url="https://github.com/gusmanb/logicanalyzer"
license=('GLP-3.0')
replaces=()
depends=()
makedepends=(git
    dotnet-sdk
    ttf-dejavu)
backup=()
options=('!strip')
install=
source=("${pkgbase}::git+${url}.git")
sha256sums=('SKIP')
noextract=()

pkgver(){
    cd "${srcdir}/${pkgbase}"
    git describe --long --tags | sed 's/^V//g;s/\([^-]*-g\)/r\1/;s/-/./g'
}

build() {
#     cd "${srcdir}/${pkgname-git}/Software/LogicAnalyzer"
#
#     if [ "$CARCH" == "aarch64" ]; then
#     msg2 "build for arm64"
#         dotnet publish  -r linux-arm64 -c Release -o artifact
#     else
#     msg2 "build for x64"
#         dotnet publish  -r linux-x64 -c Release -o artifact
#     fi

#first step, prepare the structure
    mkdir -pv  "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"
    mkdir -pv "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/CLCapture"

if [ "$CARCH" == "aarch64" ]; then
    msg2 "build for arm64"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/SerialProtocolAnalyzer"
    dotnet publish -r linux-arm64 -c Release -o artifact
    cp ./artifact/SerialProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/I2CProtocolAnalyzer"
    dotnet publish -r linux-arm64 -c Release -o artifact
    cp ./artifact/I2CProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/SPIProtocolAnalyzer"
    dotnet publish -r linux-arm64 -c Release -o artifact
    cp ./artifact/SPIProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    #third step, compile LogicAnalyzer
    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/LogicAnalyzer"
    dotnet publish -r linux-arm64 -c Release --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer"

    #fourth step, compile CLCapture
    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/CLCapture"
    dotnet publish  -r linux-arm64 -c Release -p:PublishSingleFile=true -p:PublishTrimmed=true --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/CLCapture"
else
    msg2 "build for x64"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/SerialProtocolAnalyzer"
    dotnet publish -r linux-x64 -c Release -o artifact
    cp ./artifact/SerialProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/I2CProtocolAnalyzer"
    dotnet publish -r linux-x64 -c Release -o artifact
    cp ./artifact/I2CProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/SPIProtocolAnalyzer"
    dotnet publish -r linux-x64 -c Release -o artifact
    cp ./artifact/SPIProtocolAnalyzer.dll "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer/analyzers"

    #third step, compile LogicAnalyzer
    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/LogicAnalyzer"
    dotnet publish  -r linux-x64 -c Release --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/LogicAnalyzer"

    #fourth step, compile CLCapture
    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/CLCapture"
    dotnet publish  -r linux-x64 -c Release -p:PublishSingleFile=true -p:PublishTrimmed=true --self-contained true -p:DebugType=None -p:DebugSymbols=false -o  "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/CLCapture"
fi
}

package_logicanalyzer-git() {
    provides=(${pkgname%-git})
    conflicts=(${pkgname%-git})

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/"

#     install -Dm0755 LogicAnalyzer/LogicAnalyzer "${pkgdir}/usr/bin/logicanalyzer"

    install -dm0755 ${pkgdir}/usr/share/${pkgname%-git}

    install -Dm0755 /dev/stdin  "${pkgdir}/usr/bin/${pkgname%-git}" <<EOF
export COREHOST_TRACE=1

cd /usr/share/logicanalyzer/
LC_ALL=C ./LogicAnalyzer
EOF

    cp -rv LogicAnalyzer/* ${pkgdir}/usr/share/${pkgname%-git}

    install -Dm644 "$srcdir/logicanalyzer-git/Software/LogicAnalyzer/LogicAnalyzer/Assets/Logo40.png" "$pkgdir/usr/share/icons/hicolor/256x256/apps/${pkgname%-git}.png"

    install -Dm0644 /dev/stdin "${pkgdir}/usr/share/applications/${pkgname%-git}.desktop" << EOF
[Desktop Entry]
Name=${pkgname%-git}
Name[zh_CN]=${pkgname%-git}
Comment=${pkgdesc}
#MimeType=application/x-${pkgname%-git};
Exec=${pkgname%-git}
Type=Application
Categories=Development;Tool;
Terminal=false
Icon=${pkgname%-git}
Version=1
EOF

}

package_clcapture-git() {
    provides=(${pkgname%-git})
    conflicts=(${pkgname%-git})

    cd "${srcdir}/${pkgbase}/Software/LogicAnalyzer/artifact/"

    install -Dm0755 CLCapture/CLCapture "${pkgdir}/usr/bin/clcapture"
}

```