gusmanb/logicanalyzer #80: LogicAnalyzer-5.0.0.0-osx-arm64

chienwanhuang, 2023-09-20:
I got the LogicAnalyzer run on M1 chip Macos, but I can't start up the soft, here is the message below
zsh: killed ./LogicAnalyzer

Could someone please help, thanks a lot.

gusmanb, 2023-09-20:
Most probable cause is the fact that the app is not signed and it must be removed from quarantine. Follow the [instructions at the Wiki](https://github.com/gusmanb/logicanalyzer/wiki/04---LogicAnalyzer-Software#notes-for-linux-and-macos-users) and try again.

Cheers.

shawnferry, 2023-09-27:
I'm seeing the same problem even after removing the quarantine bit either on the zip before extraction or on all the files in the extracted directory.

gusmanb, 2023-09-27:
If you already have correctly removed the directory from quarantine then it may be a problem with your zsh config, check this post as it contains some hints about what you could do: https://discussions.apple.com/thread/254910900

chienwanhuang, 2023-09-28:
I done the command "sudo xattr -r -d com.apple.quarantine {directory}",  but the problem still exsisted.  Should it must be reinstall the MACOS to solved??

gusmanb, 2023-09-28:
> I done the command "sudo xattr -r -d com.apple.quarantine {directory}", but the problem still exsisted. Should it must be reinstall the MACOS to solved??

Please, check the post, there are multiple guidances in there, reinstalling MacOS was what the user did as the last instance but it seems that he had a very corrupted system, before doing something so radical there are multiple options to check: try to set bash temporarily as your main shell, check the contents of ~/.zshrc, ~/.zshenv ~/.zprofile and ~/.zlogin for corruptions or commands that could kill the program, re-assign zsh as your sell, etc.

chienwanhuang, 2023-09-28:
which zsh
/opt/homebrew/bin/zsh

open ~/.zprofile
eval "$(/opt/homebrew/bin/brew shellenv)"
export PATH="/opt/local/bin:/opt/local/sbin:$PATH"
export MANPATH="/opt/local/share/man:$MANPATH"
PATH="/Library/Frameworks/Python.framework/Versions/3.10/bin:${PATH}"
export PATH
====

I touch a new .zshrc as below.
open .zshrc
export PATH=".:/usr/local/bin:/opt/homebrew/bin:${PATH}"
PS1="%m: %~ %% "
====

~/.zshenv & ~/.zlogin does not exist.

when I run the logicanalyzer on zsh, it shown as below
[1]    6435 killed     ./LogicAnalyzer

Then I change to sh, run the programe again, it still got killed.
Killed: 9

gusmanb, 2023-09-28:
Sorry but I can't help there, I have no M1 Mac nor I use MacOS, it seems that is a relatively common problem with MacOS and there can be many causes, there are lots of posts related to that issue, you will need to seek help on the Mac forums.

Cheers.

closed 2023-10-12 by gusmanb

mengstr, 2023-10-12:
This error interested me so I tried as well. And I have exactly the same problem, even after updating the xattr.  The linked post seems not relevant since that was an issue with just opening the zsh, not running a program thet gets killed.

Could you please make a small HelloWorld program that just prints a message and then exists with error code 0? Just to see if it's a problem with running any C# code, or if it's caused by something more advanced than a print in the code. It wouldn't surprise me if it's the usblib that is acting up.

gusmanb, 2023-10-12:
> This error interested me so I tried as well. And I have exactly the same problem, even after updating the xattr. The linked post seems not relevant since that was an issue with just opening the zsh, [...]

The thing is that other users are using the app with Apple ARM devices and the only thing they needed to do is the xattr, so it's not a problem with C#:

https://github.com/gusmanb/logicanalyzer/issues/65

Other users codesigned the application but for that you need to have an Apple developer account if I'm not wrong:

https://github.com/gusmanb/logicanalyzer/discussions/75

gusmanb, 2023-10-12:
> Ah, ok. I had no joy by just doing the xattr as in #65. But after signing the code with an ad hoc key (my real developer key expired 6 month ago) it works just fine. :-)

Is codesigning available in any mac without installing the developer tools? If that's the case then I can expand the documentation with the codesigning step.

mengstr, 2023-10-12:
Not sure... I'm currently downloading the 13GB MacOS image so I can install a clean one as a Parallels VM and see exactly which steps that are necessary in a virgin MacOS.

jdloft, 2024-03-12:
This worked for me:
1. `chmod +x osx-arm64/LogicAnalyzer`
2. `sudo xattr -r -d com.apple.quarantine osx-arm64/`
3. Make certificate in XCode (Settings > Accounts > Manager Certificates > Add (+)
4. Copy CN from certificate
5. `codesign -s "CN HERE" osx-arm64/LogicAnalyzer`