gusmanb/logicanalyzer #82: Protocol Analyzers not found on Case Sensitive Filesystems (Linux)

shawnferry, 2023-09-27:
Version: 5.0.0.0
OS: Linux

Protocol analyzers reports "None" when directory is called `Analyzers`.
Protocol analyzers reports <list> when directory is called `analyzers`.

I would propose looking for `Analyzers` instead of the lowercase version. as that appears in the release zip files. For non-case sensitive file systems (Windows and most/standard OS X) it should continue to work. 

Workaround: Manually rename directory.

gusmanb, 2023-09-27:
In fact is a problem in the package, it has been always named "analyzers" (you can check older versions), not sure why this last time it got into the package with uppercase...

In any case, to avoid problems I think the best would be to check for both possibilities, that will make it less error-prone for anyone that wants to compile it.

gusmanb, 2023-10-24:
Correction added and commited, it will be included in the next release. Meanwhile you can download the repo and build it :)

closed 2023-10-24 by gusmanb