gusmanb/logicanalyzer #87: QOL I2C start/stop/repeatstart are commonly invisible

shawnferry, 2023-10-16:
I'm trying not to pack too many things into a single issue.

As in the region highlighted in #86 even fairly low sample display doesn't always show start/repeat start and rarely shows stops.
I think it would be helpful to either always show (or optionally) show start/stop/repeat start as lines on the rendered display.

With colors is fine but possibly also with a simple geometric indicator circle/triangle/square or single letters S/T/R (to allow for color blind users and to differentiate from trigger locations etc)

My rough thought was something like this [image]

Unfortunately I don't have any C# experience and it wasn't immediately clear how I could just cram something like that into the parser.

gusmanb, 2023-10-16:
Ok, I will think something for the next update, I wanted to do a release last week but right now I'm really busy with my job, I hope to have an update in a couple of weeks.

Cheers.

shawnferry, 2024-05-05:
🎉

gusmanb, 2024-05-05:
Damn, I closed this incorrectly, still not implemented... sorry :/

gusmanb, 2024-05-11:
Ok, I think this will solve the problem.

Now the events will be always rendered, if the space is not enough for the text only the text will be hidden, and also, all the events will be shown as colored areas to the sample marker.

[image]

I expect to publish the version with this changes next week.

Cheers.

closed 2024-05-12 by gusmanb