gusmanb/logicanalyzer #95: Measure Window Blank

tibohu, 2024-03-18:
Dear Gusmanb,
there is a strange behaviour. The popup Measure Window became blank, it worked perfectly before.
Even stranger: I deleted the full app folder "LogicAnalyzer-5.0.0.0-win-x64". Unzipped the untouched download zip file again. Started the LogicAnalyzer again but Measure still not working as expected.
[image]

gusmanb, 2024-03-18:
Hi.

Did you changed your monitor or had another one connected and now you removed it?

In any case, the settings are not stored in the application folder, they go at %appdata%\LogicAnalyzer (usually "C:\Users\(your user)\AppData\Roaming\LogicAnalyzer" on Windows, "home/.config/LogicAnalyzer" on Linux), delete the AppConfig.json and that should reset all the layout settings.

Cheers.

tibohu, 2024-03-19:
Thank you!
Deleting the AppConfig.json file solved the problem.
Regarding your question, I did not change or connect to another monitor.

by the way, it would be nice if we
-could change the color of the channels
-keep the last setting in the windows
(should I place these into a newer thread?)

gusmanb, 2024-05-12:
Channel colors can now be selected and are preserved in stored captures.

Will be included in the next release.

[image]

closed 2024-05-12 by gusmanb