gusmanb/logicanalyzer #96: Parallel analyzer

mlukasek, 2024-04-06:
Thank you for your analyzer. It has quite a lot of pins, so it is useful to analyze parallel busses. It would be very useful to have simple parallel analyzer to show eg. 8 bits/signals in form of hex number.

gusmanb, 2024-05-05:
Hi.

If you can provide a capture (.lac file) I could prepare one.

For it I would need:

-Read/Write signals
-Data bus (sequential channels for the bits please :) )

Cheers.

mlukasek, 2024-05-07:
Hi,

thanks, this would be great.

I attached sample of 8080 CPU boot. There are data D0-D7 on channels 1-8, addresses A0-A11 on channels 9-20. MR (memory read) is on channel 22 and I would sample it on rising edge as the data are still valid on rising edge (the real read appears some time before it, but definitely some access time delay after the falling edge). The MW (memory write) is on channel 21 and I would sample the falling edge, as the address and data should be valid at that time.

The decoded values of adddress and data bus should be following (I also provide assembler instructions to make it more understandable):
```
0000 C3 8B 01    JMP 018BH
018B F3          DI
018C 21 6D 40    LXI H,406DH
018F F9          SPHL
0190 AF          XRA A
0191 77          MOV M,A
----> write 00 to addres 406DH (the highest 4 bits are not visible in capture)
0192 2C          INR L
0193 C2 91 01    JNZ 0191H
0191 77          MOV M,A
----> write 00 to addres 406EH (the highest 4 bits are not visible in capture)
... loop continues
```

[SAPI_JPR-1_MIKOS5_boot_LAC.zip](https://github.com/gusmanb/logicanalyzer/files/15234198/SAPI_JPR-1_MIKOS5_boot_LAC.zip)

Thank you very much!

Best regards

Martin

gusmanb, 2024-05-07:
Excellent, this weekend I will start checking it.

gusmanb, 2024-05-11:
Hi!

I think I got it working. If you tell me which OS are you using I will compile a test version so you can check it :)

[image]

Cheers.

mlukasek, 2024-05-11:
Great, thank you! I use mostly Windows 11, 64 bit.

gusmanb, 2024-05-11:
Ok, here is the version:

gusmanb, 2024-05-11:
Ok, here is the version.
https://mega.nz/file/mChQwKaQ#Bv6JkoX7tppgmKPgFv6ANXhrDkUD3mftZxxmTLGtM6o

This is based on V5.1 so you might need to update the analyzer firmware if your device is in V5.0

Now, how to use the analyzer.

It allows you to select three control signals: CS, RD and WR. Not all three are required and the behavior will change based on what you choose.

If you provide the CS line it will be used as the trigger for the sample, if it is not provided then the RD/WR signals will be used as triggers (you can even provide a single one).
If CS is provided then the RD/WR signals will be used to state which operation was carried on, if one or both of those signals are missing the non-provided ops will be shown as "V" for "value".

You can provide the same signal for multiple purposes, for example if you use the CS line and your device only has the WR signal you can use the same channel for RD and WR specifying a different edge.

The edge of the control signals is selectable so it can be configured as needed.

The protocol expects one or two buses, the Data bus is mandatory and the Address bus is optional, if no Address bus is provided the address will not be included in the analysis.

All the bus signals must be sequential channels in LSB to MSB order.

After you select the bus signals you can specify the bus width, for Data the minimum bus width is 4 bits up to 32 bits, and for address it can go from 0 bits (disabled) to 32 bits.

Also I have included a sampling offset, if it is specified it will skip n samples from the trigger condition to execute the sampling of  the data.

mlukasek, 2024-05-11:
Thanks, I tried, but I am not able to find out how to set the analyzer parameters to get correct data. It usually displays nothing, only CS seems to work somehow strange. Maybe I just do not understand the behavior. It would be great, if the analyzer could remember last settings, as you have to fill all parameters again and again now.

Should this be correct setting for my sample data (no CS, RAM is always selected)?
Channel for signal CS: MR
Channel for signal RD: MR
Channel for signal WR: MW
First channel of bus Data: D0
First channel of bus Address: A0
CS edge: Falling
RS edge: Rising
WR edge: Falling
Trigger offset (in samples): 0
Data width: 8
Address width: 12

It does nothing :( 

Thanks, 
Martin

gusmanb, 2024-05-11:
Hi.

-Do not use the CS signal, your device does not have it as you are reading the processor bus, as explained in previous post depending on the provided signals the behavior will change, provide the RD and WR signals only, these will be used as source triggers and operation definition signals.
-RS edge in your capture is falling, not rising
-In your capture there are some cycles of delay between the RD signal going low and the data being active (this is normal, is the setup time for the RAM or ROM) so give it 3 or 4 cycles of offset to the trigger.

With these it should work as expected.

Also, once you get the options correct you can use the "repeat last analysis" and you don't need to enter the parameters again.

Cheers.

mlukasek, 2024-05-11:
Oh, I see, thank you. Now I understand it better and it works! The offset is clever. Maybe it could be different for read and write, as read has memory read access time, while write has data prepared before the edge. In any case it would be nice if it could remember last settings. Can the bus be split between two analyzers?

I also found another reason why I did not see anything and considered it strange behaving. When "samples in screen" make the pulse small, that data visibly does not fit, it displays nothing. Maybe it could display just the "hexagon" without text to make clear that there is something decoded, but not visible.

Thank you very much!

gusmanb, 2024-05-11:
Excellent :)

Ok, I will split the offset in two, for read and write, makes sense.

About the size... hmm, what you propose will also solve other related post, I will check if I can do it.

Cheers.

gusmanb, 2024-05-11:
Oh, also, yes, you can split the channels between multiple analyzers, the only requirement is that the channel indexes for the buses are sequential, if you use the chaining of the analyzer you can connect up to five analyzers and it gives a total ammount of 120 channels, there can be a sample of delay between the master analyzer and the secondary ones but with the offset you can adjust it.

gusmanb, 2024-05-11:
Ok, got the offsets splitted, the settings are preserved, the events are now always rendered and finally I have added the events as a colored region to the sample marker, so you can see always all the events that happened even if the channels aren't in screen.

[image]

Tomorrow I will compile the version and send it to you.

Cheers.

mlukasek, 2024-05-12:
Perfect, looks great, thank you!

gusmanb, 2024-05-12:
Here it is:

https://mega.nz/file/jSZUxDab#AgwE_6eDr3jveww2gw0t2relk2tQN_AG3NftfgN3RCc

gusmanb, 2024-05-12:
I'm going to close the issue as completed, if you find any problem feel free to reopen it or create a new one.

closed 2024-05-12 by gusmanb

mlukasek, 2024-05-13:
Works great, thank you!