gusmanb/logicanalyzer #97: SDQ Protocol Analyser

ElectroBoy404NotFound, 2024-05-01:
Hi there,

It would be a helpful addition if an SDQ Protocol Analyser is implemented. I'm working on reverse engineering Apple's Lightning protocol based on SDQ. It would be helpful if I could view the hexadecimal values instead of the raw voltage signals and decode them by hand.

gusmanb, 2024-05-01:
If you provide a capture of the protocol I could try to implement it, this is a protocol that I never used...
From the docs I think that with a capture at 10x sample rate and the bitrate could be enough.

gusmanb, 2024-05-02:
Hi.

I need it in the .lac format generated by the LogicAnalizer app as it contains a lot more information (File->Save).
Also I need to know the bitrate of the coms.

Cheers.

gusmanb, 2024-05-11:
Hi.

Could you provide the info please? I'm working in adding some analyzers and would be good to add all of them at once.

Cheers.

ElectroBoy404NotFound, 2024-05-13:
> Hi.
> 
> Could you provide the info please? I'm working in adding some analyzers and would be good to add all of them at once.
> 
> Cheers.

Hi there,

Sorry, I actually accidentally killed the device; I have another device which I'll take a sample once I get a lightning port.

closed 2024-10-07 by gusmanb

ElectroBoy404NotFound, 2024-10-08:
I completly forgot about this... Sorry for making you wait!

I will open a new issue once I have the lightning recordings and all

gusmanb, 2024-10-08:
> I completly forgot about this... Sorry for making you wait!
> 
> I will open a new issue once I have the lightning recordings and all

Hi.

No worries, the new version includes all the analyzers from Sigrok so it is already implemented.

Cheers.

ElectroBoy404NotFound, 2024-10-08:
> > I completly forgot about this... Sorry for making you wait!
> > I will open a new issue once I have the lightning recordings and all
> 
> Hi.
> 
> No worries, the new version includes all the analyzers from Sigrok so it is already implemented.
> 
> Cheers.

Oh ok, thank you!