gusmanb/logicanalyzer #98: Issue when analyzing SPI protocol in mode 2

miguelagve, 2024-05-04:
Hello, 
I was trying to use the logic analyzer software (v5.0.0.0) for some electronic projects, and the output I'm getting from the SPI analyzer doesn't match the actual message sent.
In the analyzer I set the CS to CH4, CK to CH1, MISO in CH7, MOSI in CH6, Shift order as Right to Left, CPOL to 1 and CPHA to 0.
The actual data I'm sending is 0x060F25, but the software is not giving the right output. It seems to me like it is missing the first falling edge, and thus causing 1 bit left shift in the data. At the end, it seems like it read one bit too many as if it skipped one of the actual bits earlier.
I'm using a stock RPI Pico with the 5.0.0.0 firmware as well
I would appreciate some insight about this issue. Here I attach a screenshot, as I cannot upload the .lac file to GitHub.
Thank you 
[image]

gusmanb, 2024-05-04:
can you save the capture and send it to me so I can test it?

miguelagve, 2024-05-04:
Yep, here you go
[DAC2_060F35_write.zip](https://github.com/gusmanb/logicanalyzer/files/15210514/DAC2_060F35_write.zip)

gusmanb, 2024-05-04:
Ok, I think I got it solved, I had somethings a bit messed up, try this new version of the analyzer.
[SPIProtocolAnalyzer.zip](https://github.com/gusmanb/logicanalyzer/files/15210634/SPIProtocolAnalyzer.zip)
Replace the DLL from the "Analyzers" folder.

Also, I think you made a typo, the message is 0x060f35, not 0x060f25.

Finally, a recommendation, use at least a sampling frequency of four times the clock of the signal, you want to have at least two samples per clock phase, if there is any desinc or glitch you may miss events.

gusmanb, 2024-05-04:
Uhmmm... Is possible that the DLL does not work for your version, I've been working in a new one and the DLL has some changes and might be incompatible...

Here you can download a preview of the new version, it includes some changes to the configuration management, remembering last network settings and also it will show in the analyzers the channel names instead of the numbers.

https://mega.nz/file/eOBDAJzL#I3Oe37QCMbzgtjiZ96BTyKkCcz5XMDWsIpfsKyKbzrI

gusmanb, 2024-05-05:
I have created a new release and includes the corrected SPI analyzer. If you can please confirm if it works for you.

Cheers.

miguelagve, 2024-05-05:
The dll does seem to work just fine on the release version, at least for this capture.
I was going to test the preview as well now, but I assume I can go ahead and use v5.1.0.0 that already has the changes.
Thank you!!
[image]

gusmanb, 2024-05-05:
I was writting at the same time you were posting 👍 

I will close then the issue.

Cheers!

closed 2024-05-05 by gusmanb