gusmanb/logicanalyzer #99: Multidevice captur

mlukasek, 2024-05-13:
Hello, I try Multidevice capture, but without success. The trigger seems to work OK, but there is almost nothing displayed on main screen. There seem to be signals captured from first device (MASTER) in signal preview, but not form the second one (the SLAVE). If I swap devices it behaves the same, only MASTER visible in signal preview and almost nothing on main screen. Both analyzers work if used alone.

I use Firmware-LogicAnalyzer-5.1.0.0-W-WiFi.uf2 and have the EXT_CHAIN and GND connected. Using COM10 and COM11, no WiFi. Latest hardware v4.0.

Did I do anything wrong or is it some issue?

Thank you, Martin.
[image]
[image]
[image]

mlukasek, 2024-05-15:
Thank you, I understand, I do not care much now about the trigger or preview or what I see. I want to just capture from both analyzers. I will set the correct signals, channels later ;-) I found, that the problem is when I select all 24 channels from Master then the channels from Slave are not captured. I tried only few from Master and it works. It even works with 23 channnels, but 24 is the problem. See attached capture files.
[capture_Multi_Martin.zip](https://github.com/gusmanb/logicanalyzer/files/15318887/capture_Multi_Martin.zip)

gusmanb, 2024-05-15:
Thanks for the capture, I will check it.

Cheers.

gusmanb, 2024-05-15:
Ok, this one has been easy, it was a bumb error in the rendering routine that went beyond the channel color arrays.

[image]

I will check the CS problem and will send you a version with both fixes.

Cheers.

gusmanb, 2024-05-15:
Ok, I think I got all sorted.

The channel viewer (not the preview) should show you now all the captured channels.
For the CS selection, if you change it to "< NONE >" it will deselect correctly the channel.

Here is the new version: https://mega.nz/file/rPAFRJ6S#9qa-LDqqxHqw8ppV6v3puSrMToAYdSWzkLvB2dhXwXU

mlukasek, 2024-05-16:
Great, thank you. It works OK, I can select < NONE > to disable CS. I just do not understand the CS behavior. If I select some signal, then it analyzes the CS and stops analyze RD and WR. How does it behave?

gusmanb, 2024-05-16:
Excellent.

There can be a +-1 sample of difference but the software tries to correct it and most of the times it does a good job, only at the highest speeds it is noticeable (and even in that case you can shift the channels to align it :) )

closed 2024-05-16 by gusmanb

mlukasek, 2024-05-16:
I tried it even on 100 MHz and it was perfectly aligned. Good job! Thank you!