gusmanb/logicanalyzer #101: Parallel analyzer - CS issue

mlukasek, 2024-05-15:
I found small bug in Parallel analyzer. When I select CS to anyhting, it is not possible to deselect it again. There is option <None>, but it does not help to select <None>. I have to close and reopen whole application. LAC file attached.

I wanted to reopen the Parallel Analyzer issue thread, but there is no button for reopen :( Sorry.

[SAPI_JPR-1_MIKOS5_boot_CS-None.zip](https://github.com/gusmanb/logicanalyzer/files/15319059/SAPI_JPR-1_MIKOS5_boot_CS-None.zip)

gusmanb, 2024-05-15:
Ok, thanks a lot for the report.

I will check the capture and the analyzer.

Cheers.

mlukasek, 2024-05-16:
Great, thank you. It works OK, I can select < NONE > to disable CS. I just do not understand the CS behavior. If I select some signal, then it analyzes the CS and stops analyze RD and WR. How does it behave?

gusmanb, 2024-05-16:
Perfect :)

If you select the "CS" line then it controls when the device becomes active, and thus there is an operation, the "RD" and "WR" signals are only used to determine which operation happens (and they aren't required if you don't have them), you will see all the operations in the CS line but with "Read: xxx", "Write: xxx" or "V: xxx", if there is an operation and no op line was active or it was not provided then it will show "V: xxx" as it can't determine which one happened.

Think for example about a ROM bus, it does not really have the read/write lines, only the CS one, so in that case you will select the CS and it will show all the ops as "V: xxx" on it.

closed 2024-05-16 by gusmanb