gusmanb/logicanalyzer #106: Wiki: Add board thickness and BOMs

cefiar, 2024-05-30:
Notice that the wiki hardware page doesn't have details on if the PCB thickness makes a difference, or what thicknesses have been tested. Can we assume that the default 1.6mm thickness PCB's are suitable?

I also note that there isn't a BOM at all for either the main Logic Analyzer board or the Level Shifter.

PS: Happy to actually add the PCB thickness and the BOM in to the wiki, but thought I'd ask first before just making the change.

mengstr, 2024-05-30:
The wiki doesn't have details of what colors of the soldermask  that are tested either ;-)  

(Green soldermask is usually the one that can produce the narrowest dam sizes so it could have an impact in some cases - just as the board thickness would affect controlled impedance traces (none of them on this board))

cefiar, 2024-05-31:
Re: BOM

This is what I've come up with. If this seems correct, feel free to grab bits of it/adapt as appropriate. The tables are all done in Github Markdown. Will note that all the listed items are available from Digikey (where I'm planning to order from), and I can add the Digikey codes to the tables as well if that is desirable.

## BOM

### Logic Analyzer

| Qty | Item Detail | Example supplier code |
| --- | --- | --- |
| 1 | Raspberry Pi Pico (or Pico W) | Raspberry Pi SC0915 (Pico) or SC0918 (Pico W) |
| 2 | 20 way straight pin headers 2.54mm (0.100") pitch | Adam Tech PH1-20-UA |
| 1 | 1N4148 General Purpose Diode, SOD80/Mini-MELF package | Onsemi LL4148 |
| 1 | 100K Resistor, 1%, 1/10W, SMD 0603 package | Vishay Dale CRCW0603100KFKEA |
| 1 | 30 pin dual row right angle header socket (2x15 pin) | Sullins Connector Solutions PPPC152LJBN-RC |

If you want to daisy-chain boards, also add the following:

| Qty | Item Detail | Example supplier code |
| --- | --- | --- |
| 1 | 3 way right angle header socket (1x3) | Sullins Connector Solutions PPPC031LGBN-RC |
| 1 | 3 way right angle header pins (1x3) | Adam Tech PH1RA-03-UA |

### Level Shifter

| Qty | Item Detail | Example supplier code |
| --- | --- | --- |
| 1 | 30 pin dual row right angle header socket (2x15 pin) | Sullins Connector Solutions PPPC152LJBN-RC |
| 1 | 30 pin dual row right angle header pins (2x15 pin) | Adam Tech PH2RA-30-UA |
| 12 | 0.1uF (100nF) Ceramic Capacitor 16V 10% SMD 0603 package | YAGEO CC0603KRX7R7BB104 |
| 6 | Texas Instruments TXU0104 IC Translator Unidirectional IC 14-TSSOP package | Texas Instruments TXU0104PWR |

> [!NOTE]
> 
> 20 way headers are not necessary if you buy a Pico H/Pico WH, as the H suffix means they have the headers already soldered on. However, the cost difference may be more than just buying the headers would be and soldering them on yourself. You can also buy the headers in longer break-away lengths that may be cheaper (eg: a single 40 pin length snapped in half, etc).

gusmanb, 2024-05-31:
Thanks a lot for the collaboration :)

I'm going to add this to the hardware section of the Wiki.

gusmanb, 2024-05-31:
Completed, now you can find the BOM in the [hardware section](https://github.com/gusmanb/logicanalyzer/wiki/02---LogicAnalyzer-Hardware).

I also have added a note about the thickness, I always use 1.6mm.

closed 2024-05-31 by gusmanb