gusmanb/logicanalyzer #107: Software for osx-arm64 not running on Apple M3 Pro Silicon

opcodex64, 2024-05-31:
Hello gusmanb,

first, I want to thank you and all contributors for creating this beautiful project. I am just happy to have a logic analyzer software for OSX which is not crashing that often....

Second, I would like to share the following experience:

**System:** Apple M3 Pro (Apple ARM64)
**OSX:** Sonoma 14.4.1

**LogicAnalyzer and ClCapture Release osx-arm64:**  Not running. According terminal output, the process is immediately killed.
**LogicAnalyzer Release osx-x64:**  Running fine. (I did not try ClCapture, but I assume it will work)

Kind regards,

opcodex64, 2024-05-31:
Hello gusmanb,

yes, before trying I have read the notes and I considered the both commands, where by the chmod is mandatory for making the binary to be executable. 

Is there any way to get "more" output from the terminal by executing the software such as a verbose mode?

opcodex64, 2024-05-31:
Thank you for the link. 

I can confirm that executing 
**codesign --force --deep -s - ./LogicAnalyzer**
solved the issue. 

Now I can run also the osx-arm64 release on M3 Silicon. 

Thanks for your help.

PS: Since I am trying to make a customized Analyzer for a certain ADC, I have downloaded the entire  LogicAnalyzer Visual Studio Solution as suggested in the documentation. I had never worked before with Visual Studio for MAC, but it is working like a charm. The LogicAnalyzer Solution could be imported easily and is also compiled with no errors on my apple m3 machine. Further, the documentation for implementing a customized analyzer is also very useful. Thanks for such a good documentation.

Kind regards,

gusmanb, 2024-05-31:
Glad it worked.

If you get the analyzer for the ADC running feel free to create a pull request and I will incorporate it to the existing ones.
Also if you need help implementing it do not hesitate in asking and I will try to help as much as I can.

Cheers.

closed 2024-05-31 by gusmanb

opcodex64, 2024-05-31:
Sure, I can do.

Is it possible to import a simulated output signal of the ADC? It would speed up the developing.