gusmanb/logicanalyzer #108: Saved Project or Workspace settings

IntergalacticMicrosystems, 2024-06-02:
Any plans on having a way to save all settings to a file?

Like:
- Channel names
- Multidevice setup (including IPs)
- Analyzer channel selections

I'm working on an adapter to use ISA cards in a Victor 9000 - it would be ideal to run 3 sessions of 2 or 3 Picos each.
One on the Victor bus, one looking at the adapted ISA signals, and one on a real PC ISA bus.

It gets a bit tedious re-setting things up.

I'm willing to work on this if it's not already a plan, and anyone is interested.

IntergalacticMicrosystems, 2024-06-04:
I've started work on this in a fork -
[https://github.com/IntergalacticMicrosystems/logicanalyzer](https://github.com/IntergalacticMicrosystems/logicanalyzer)
Basic Workspace saving / loading working - currently just the device type and settings.
Better persistence during the session also - for open/close of multidevice.

gusmanb, 2024-06-06:
Hi.

Thanks for your interest. I have checked your branch and only have a comment, with your changes the last network setting is not preserved and forces the user to always save a workspace in order to have these remembered, it would be nice to have both possibilities :)

Cheers.

IntergalacticMicrosystems, 2024-06-06:
That makes sense, if blank try and pull from json file like you had before.
I'll work on it when I get a chance.
Thanks!

joep, 2025-07-14:
Hi everyone,

Just got my order from pcbway, installed a pico and started playing. When using the software, almost the first thing I did was trying to figure out how to save & reload the capture setups. Went to the documentation, didn't see anything, so I came here and found this.

Have I missed something in the documentation perhaps or has this feature basically stalled out?

Terrific work, btw.

gusmanb, 2025-07-14:
Profiles are coming in V6.5, they're on the works, but for now the program stores the last capture settings that you have used with the analyzer.

joep, 2025-07-14:
Thanks for the quick reply. Since I hadn't restarted the program yet, I did not realize it saved the last capture settings si I'm happy with that. Looking forward to 6.5! And thanks for the great work.