gusmanb/logicanalyzer #110: Collapse signals to less lines

IntergalacticMicrosystems, 2024-06-02:
Any plans?

Example:
I'm working on an ISA bus protocol analyzer - would be nice to collapse the address bus down to the 1 line with just the hex address shown.

Last idea for now :)

Thanks, Matt

gusmanb, 2024-06-06:
I have read both comments and I think I could do something, I can create a view options dialog where you could select which channels are visible and their order.

If it is ok fdor you I would close both issues and create a new one with this change and once I get a bit of time would start working on it, latelly my work is absorbing me 100% of the time :/

Cheers.

IntergalacticMicrosystems, 2024-06-06:
Sure - works for me!
No worries, I'll give it a try too - I was just checking - if you already had something in the works, no sense duplicating effort.

gusmanb, 2024-06-06:
Merging both issues at https://github.com/gusmanb/logicanalyzer/issues/112

closed 2024-06-06 by gusmanb