gusmanb/logicanalyzer #116: i succeeded in compiling / running with vscodium, howto add to the README?

cdwijs, 2024-07-17:
I've just succeeded in compiling / running Logic in vscodium. Can the below steps be included in the README?

1) Install dotnet 7. I used this version on my Windows 10 machine:
https://dotnet.microsoft.com/en-us/download/dotnet
https://dotnet.microsoft.com/en-us/download/dotnet/7.0
https://dotnet.microsoft.com/en-us/download/dotnet/thank-you/sdk-7.0.410-windows-x64-installer
https://download.visualstudio.microsoft.com/download/pr/6f7abf5c-3f6d-43cc-8f3c-700c27d4976b/b7a3b806505c95c7095ca1e8c057e987/dotnet-sdk-7.0.410-win-x64.exe

2) download this extension. This is an alternative to the Microsoft C# Dev Kit that can only be installed on Visual Studio Code
https://open-vsx.org/extension/muhammad-sammy/csharp
https://github.com/muhammadsammy/free-vscode-csharp/releases/tag/2.34.12
https://github.com/muhammadsammy/free-vscode-csharp/releases/download/2.34.12/csharp-platform-neutral-2.34.12.vsix

2) Open vscodium, then install the extension "Avalonia for VSCode"

3) Open vscodium, choose -> view -> command palette -> Extensions: Install from VSIX -> csharp-platform-neutral-2.34.12.vsix
This will also install these dependencies:
OmniSharp,
.NET Core Debugger
Razor Language Server
Razor Language Server for OmniSharp

4) Click on "run and debug", click the green arrow next to .NET Core Lanch (console). Now logicAnalyser is compiled, and then runs.

cdwijs, 2024-07-17:
I missed a step that only has to be done once:
5: Click on "run and debug",click the blue button "Run and Debug", At the top, choose ".NET5+ and .NET CORE". Then choose "LogicAnalyser". now this json file is created:

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": ".NET Core Launch (console)",
            "type": "coreclr",
            "request": "launch",
            "preLaunchTask": "build",
            "program": "${workspaceFolder}/Software/LogicAnalyzer/LogicAnalyzer/bin/Debug/net7.0/LogicAnalyzer.dll",
            "args": [],
            "cwd": "${workspaceFolder}/Software/LogicAnalyzer/LogicAnalyzer",
            "console": "internalConsole",
            "stopAtEntry": false
        },
        {
            "name": ".NET Core Attach",
            "type": "coreclr",
            "request": "attach"
        }
    ]
}

6) Click on "run and debug", click the green arrow next to .NET Core Launch (console). Now LogicAnalyser is compiled, and Launched.

gusmanb, 2024-07-21:
Sorry but the only officially supported platform to compile the software is Visual Studio. Adding info about compiling in other IDEs would mean to mantain that info updated and cope with the problems that those other may bring.

closed 2024-07-21 by gusmanb

cdwijs, 2024-07-21:
> Sorry but the only officially supported platform to compile the software is Visual Studio. Adding info about compiling in other IDEs would mean to mantain that info updated and cope with the problems that those other may bring.

While I can respect that standpoint, that's not reflected in the wiki* There it says this
"I use [Visual Studio](https://visualstudio.microsoft.com/es/#vs-section) to create it, but you can build it using [Visual Studio Code](https://visualstudio.microsoft.com/es/#vscode-section). I encourage you to use Visual Studio as it's more more mature and full of functionalities not available on VS Code, but hey, if you don't like it or don't have enough space to install it feel free to use VS Code."

Also, how can one compile Logic Analyser on mac or Linux? Visual studio is not available on those platforms.

https://github.com/gusmanb/logicanalyzer/wiki/04---LogicAnalyzer-Software

gusmanb, 2024-07-21:
> While I can respect that standpoint, that's not reflected in the wiki* There it says this "I use [Visual Studio](https://visualstudio.microsoft.com/es/#vs-section) to create it, but you can build it [...]

You have answered yourself to both questions :)

I don't give support to VS Code, but if you want, go for it :)

Also, if you want to compile under Linux or Mac you can go with VS Code, it works on all platforms, but I don't officially support it, only compiling under Windows is officially supported.