gusmanb/logicanalyzer #117: Support for the Protocol decoders from sigrok?

cdwijs, 2024-07-18:
In logic Analyzer, the I2c, parallel, serial and SPI decoders are implemented as c# programs. This will run very fast, as they are compiled. 

In Sigrok, Protocol decoders [1] are implemented in python. This is slower, but there are a lot more of them. They can also stack, so I could first use a serial decoder, stack a modbus decoder on top, and then see what my inverter is saying.

Is it feasible to support the python protocol decoders from Sigrok In logic Analyzer?

[1] https://sigrok.org/wiki/Protocol_decoders

gusmanb, 2024-07-21:
Hmmm, I will check something, not to support the analyzers directly but create an analyzer that uses the sigrok library to use the analyzers, that may be feasible. This summer when I get vacations will check if it is possible.

gusmanb, 2024-08-18:
Check [this](https://github.com/gusmanb/logicanalyzer/discussions/127), I think you will like it ;)

gusmanb, 2024-10-07:
Implemented in V6.0

closed 2024-10-07 by gusmanb