gusmanb/logicanalyzer #124: Logic Analyser uses .net 7 that is unsupported. 6 and 8 are supported.

cdwijs, 2024-07-21:
I had to install .net 7 to be able to compile Logic Analyser * .NET 7 is no longer supported:
[image]

Is there a technical reason why version 8 can't be used?

* https://github.com/gusmanb/logicanalyzer/issues/116

gusmanb, 2024-07-21:
I need to check if AvaloniaUI has been updated to .net 8.0, also if you check the last release of the project is on May 5, before it ran out of support.

If avalonia has been updated to support .net 8 (most probably yes) I will update the framwework version.

miguelagve, 2024-08-08:
As far as I can tell, I should be supported
[https://docs.avaloniaui.net/docs/faq#what-versions-of-net-can-i-use](url)

gusmanb, 2024-08-18:
Next release will come with the latest version of Avalonia and .net 8, I have already updated everything locally :)

closed 2024-10-07 by gusmanb