gusmanb/logicanalyzer #133: Some pretrigger data missing after capture

jeroentaverne-dn, 2024-09-03:
I did a capture on an continuous I2S signal (find attached). But every time I capture, most of the pre-trigger data is missing. I used version v5.1 of firmware and application. What could be the reason of this? I also want to ask if it is possible to show the trigger moment to 0, and pre-trigger samples to negative values. Also it would be nice if zooming with mouse wheel is supported and the center of the screen is the zoom reference. Thx!
[i2s.zip](https://github.com/user-attachments/files/16846623/i2s.zip)

gusmanb, 2024-09-03:
From the capture I would say that the trigger condition is already true when the capture starts.

The analyzer has a ring buffer that is initialized to zero when the capture starts, then it starts sampling on it the inputs and when the trigger condition becomes true it starts counting samples until it reaches the post samples ammount.
If the trigger is true when the capture starts the buffer has no pre-trigger data and it will be sent empty.

In V6 I tested the zooming over the sample viewer but it has a problem, when there are too many channels to fit in the screen the wheel stops controlling the zoom and controls the position of the channel/sample viewer. What I have implemented is to be able to use the scrollwheel over the sample selector, is not the same but at least it works with the wheel.

About being negative, I will check it :)

Cheers.

jeroentaverne, 2024-09-04:
Thanks for your great explanation, it makes sense now. The trigger happens soon indeed. Would it perhaps be possible to hide the empty data in the gui? So far a very useful project! Thanks!

gusmanb, 2024-09-04:
Will check if possible in next version :)

closed 2024-09-04 by gusmanb