gusmanb/logicanalyzer #136: PCB production files

drtonis, 2024-09-09:
Would it be possible to add PCB production files? Gerber / Schematics / BOM / Pick and place files would be nice too.
In the moment you have Kickad files, however I dont have the program and it's quite challengint to figure out what I need.
Thanks!

gusmanb, 2024-09-09:
Hi.

I don't add the gerber files because each manufacturer may need these with different specs, KiCad is 100% free and installing it and exporting the files is very simple, you can [download it here](https://www.kicad.org/download/) and [here is a link](https://jlcpcb.com/help/article/how-to-generate-gerber-and-drill-files-in-kicad-7) on how to export them for JLCPCB.

Regarding the BOM [it is included in the wiki](https://github.com/gusmanb/logicanalyzer/wiki/02---LogicAnalyzer-Hardware#bom).

WIth V6.0 I will add the schematics and PCB files in PDF.

If you have problems exporting the file drop a message here with the specs and I can export it for you. 

Cheers.

arekm, 2024-12-10:
Maybe add project to something like https://www.pcbway.com/project/shareproject/ and it will be easy for people to order PCB or complete assembly? Some form of support for the project is included - AFAIK you will get 10% of each order.

jlcpcb unfortunately doesn't have such thing - https://jlcpcb.com/help/answers/detail/555-Sharing-my-PCB-design-so-others-can-order

No idea about other PCB manufacturers.

It would make much easier to order.

ToxxikHalo, 2024-12-11:
i agree with arekm that sharing the project on pcbway would go a looong way to help out the community actually getting their hands on this awesome device... i would literally be ordering ac couple of them right now but i found this while trying to find out if were were prebuilt devices available or where i could obtain a kit if not... ill still do whatever is necassary to get one or 2 of them but just ordering themon pcbway would open up doors to other people who arent sure how to go about it etc.... 
thanks for your time and awesome project! i cant wait to get to try one out!!!

gusmanb, 2024-12-11:
> i agree with arekm that sharing the project on pcbway would go a looong way to help out the community actually getting their hands on this awesome device... i would literally be ordering ac couple [...]

I'm on the process of creating a website were to order preasembled analyzers. For now you can send me an email at logicanalyzerorders@gmail.com, I'm sending regularly batches.

Willem-Bever, 2024-12-11:
Hi Augustin,

I found your project true YouTube.
I came across a video from little happy diodes.
I think you design a excellent product.
I'm shortly for my retirement, but the last 20 years I was (still) working as an automotive engineer. Could you please inform me when your latest version is ready for sale.

BR
Wim (Venlo, Holland)

⁣BlueMail voor Android downloaden ​

gusmanb, 2024-12-11:
> Hi Augustin, I found your project true YouTube. I came across a video from little happy diodes. I think you design a excellent product. I'm shortly for my retirement, but the last 20 years I was [...]

Hi!

Send me an email to logicanalizerorders@gmail.com and will add you to the delivery list, once I get a new batch then I will notify you.

gusmanb, 2025-02-01:
Production files added to the release artifacts.

closed 2025-02-01 by gusmanb