gusmanb/logicanalyzer #141: Serial Analysis crashs the 64bit Windows, V5.1.0.0 application

pepperm2, 2024-09-15:
I have captured a stream of 2400 baud serial data and saved it. If I no do a protocol analysis using the Serial analyser the app sits ther for a moment then crashes after setting the Serial Analyser settings and clicking accept. This happens on two computers and if I have an attached analyser over USB or WiFi, or even if I don't attach an analyser and just Open a previously saved file. I have attached the sample file.

[serial capture.zip](https://github.com/user-attachments/files/17006060/serial.capture.zip)

gusmanb, 2024-09-15:
Hi.

The current analyzers are being deprecated in favor of a new analysis system so no bug fix is going to be done for the current version.

If you want to test version 6.0 [here is a link to it](https://mega.nz/file/TbBF0bqI#57Lsw0z6G5ymlvjCQewEgX_MvRlZ0ANrWt_qa3-pfMU) (work in progress). It contains over 100 protocol analyzers and many improvements.

You will need to upgrade the firmware with [these new versions](https://mega.nz/file/3SZCDQzD#krl3m4G7HKwNuCi6rsmw_yeyi9CK_usz6gGNyXah-uY) and install Python 3.7 or higher. There are two sets of firmwares included, the turbo ones allow to capture up to 200Mhz but beware that adds a lot of stress to the CPU because of the overclocking and overvoltage, so if you use it is under your own responsability (Pico W is not supported) 😄 

Cheers.

pepperm2, 2024-09-15:
V6.0 looks great, keep up the good work

closed 2024-09-15 by gusmanb

gusmanb, 2024-09-18:
> Any chance of getting a release of v6.0 for us Linux (Ubuntu 24) users?

Check https://github.com/gusmanb/logicanalyzer/discussions/144