gusmanb/logicanalyzer #142: Pico 2 and the 2350, floating pin issue

Multithought, 2024-09-16:
Hello, 

I came across your project on a YouTube channel.  But I saw you were porting to pico2 and stopped due to the latch-up issue

Re-read the errata.   The issue doesn’t affect your 24 input channels.  It’s not an issue.   In your device the level shifters are driving the inputs of the RP active low or high.  The issue only occurs when the pins are weakly driven low or floating.. open collector etc.   with the active, driven outputs of the level shifters, you will not see the latch-up issue.  To be extra sure if you want just turn on the pull up internal to the 2350.  

I have already made and delivered to market, a product based on the RP2350A2 (30,000 of them)    It’s not a problem in this, and indeed most situations.  

-Matt

gusmanb, 2024-09-16:
Hi. 

I've been involved in the case since the beginning in the Raspberry forums and Github. It does affect, I already diagnosed it and for now it is stopped. 

The device can be used without the shifters, the most basic form of the analyzer is the pico barebones, if anyone uses it like that the pico will mess up the analyzed signals leading to unexpected results.

There are [evidences on the E9 thread on Github](https://github.com/raspberrypi/pico-feedback/issues/401#issuecomment-2328979371) that a signal connected to an RP2350's GPIO is distorted and that's something that a logic analyzer cannot afford.

Cheers.

closed 2024-09-16 by gusmanb

gusmanb, 2024-12-12:
> @gusmanb does that mean that we can't use the 2350 (or at least some of its GPIOs) at all or that additional circuitry is required to compensate for it? I didn't quite understand that part in the video either
> 
> Maybe you can add a note on the readme?

There is a full section explaining it in the readme that is the main page of the repository: https://github.com/gusmanb/logicanalyzer?tab=readme-ov-file#good-news

The new design mitigates the problems to a point that should not be a problem, the trigger problem is corrected by adding pull-downs and the inputs also by making the level shifters not dettachable.

I observed a minimum skew that I think that is still caused by the Pico 2 but it happens only at 400Mhz so is not really a problem.

dzervas, 2024-12-12:
damn my eyes stopped working :)

thank you!

MacGyver4B, 2025-08-04:
As there is a new Revision (A4) of RP2350 and RP2354 the Pico 2 is now working fine and does not need Voltage Translators for %V operation, can that be confirmed?
https://www.raspberrypi.com/news/rp2350-a4-rp2354-and-a-new-hacking-challenge