gusmanb/logicanalyzer #143: higher speed

startcapture, 2024-09-17:
Hi very well implemented and presented project. I am looking at faster sampling and I saw in the code, 200Mhz, would I be correct this would be all I need to do?

1. remove the frequency check in CLCapture
2. remove the frequency check in startCaptureFast(). LogicAnalyzer.pio
3. set_sys_clock_khz(200000, true) to 400000. LogicAnalyzer.c

I haven't tried it, no pico development enviroment yet

gusmanb, 2024-09-17:
No, it is not all, but all is already implemented in the upcomming V6.0 😄 

In [this response](https://github.com/gusmanb/logicanalyzer/issues/141) you can download the new firmwares and software.

startcapture, 2024-09-18:
Nice, Pico 2 as well.

Just curious, having 400Mhz clock speed on SAMPLE PIO and TRIGGER PIO, I imagine this would trigger check every 2nd input sample but would allow 400Mhz sample rate. From your knowledge would this work? or does it mess with the sample count?
reason being, I'm interested in propagation delay on UART and SPI so more speed is good.

Idea, raise to 1.3 volts and sys clock to 400Mhz after capture initialise but before capture start, reduce back to 200Mhz in capture complete IRQ, Only capture at maximum speed then slow down for the rest.

Anyway keep up the good work.

gusmanb, 2024-09-18:
> Nice, Pico 2 as well.
>
> Just curious, having 400Mhz clock speed on SAMPLE PIO and TRIGGER PIO, I imagine this would trigger check every 2nd input sample but would allow 400Mhz sample rate. From your [...]

Hi.

Do not use the Pico 2, I have prepared the code in the event that I find a way to make it work properly or there is a revision without the GPIO latching, but in its current state I have seen very messed up results, at high speeds the captures were duplicated sometimes when a high pulse was sent and I assume it's because the latch lasts more than a couple of nanoseconds and then it reads twice a high level, I'm waiting for new boards that might allow it to work but for now it is not supported.

About the PIO machines, they are already running at twice the speed you capture, so when you capture at 200Msps they are running effectively at 400Mhz. The tightes loop that you can achieve having some kind of control to know when to stop capturing is two instructions, so it then becomes 200Msps.

Finally, if I recall it right I did some tests when I started developing the device that changed the clock after the USB was initialized and it messed up the communications, it was a long ago so it could be other things, I will retest again and if it works I will add it.

Cheers.

startcapture, 2024-09-18:
OK, thanks for the info on the Pico 2, I see 600+Mhz overclock 

Ditto the PIO machines. I see some code to input through GPIO wth DMA that dosen't do pre trigger/ringbuffer DMA sampling and is triggered on a single gpio change, would you know if that would miss any initial samples, ie. DMA start delay?

Oh I was going to ask if UART communications was (to/from Pi 4 to pico) an option instead of USB

gusmanb, 2024-09-18:
> OK, thanks for the info on the Pico 2, I see 600+Mhz overclock
>
> Ditto the PIO machines. I see some code to input through GPIO wth DMA that dosen't do pre trigger/ringbuffer DMA sampling and is [...]

Not sure about which code you refer so I cannot help there...

About the UART, no, it is not an option, the pins are already used by the analyzer.

Cheers.

startcapture, 2024-09-19:
Thanks, I will include this for the search engines, and to help raise awareness.

I used the pico logic analyser first time last weekend for a dead Asrock Z170 motherboard, BIOS chip corrupt. removed the chip into a bread board but wouln't recognise it with flashrom program. Downloaded this project, flashed pico and found clock signal connection problem, fixed it, flashed the BIOS chip, motherboard booting ok, fixed. During this week the logic analyser sample rate has doubled to 200Mhz (experimental) and the included software is easy to use and sensible.
1. download prebuilt version and flash firmware to the Pico
2. make the program executable: $ -rwxrwxrwx 1 ---- ----  201304 May  5 13:53 LogicAnalyzer
3. run: $ ./LogicAnalyzer

Question. I see Linux *.so library files, are the Windows *.dll needed?

code I'm referring to is the simple example logic analyser in Pico SDK. Are you sending raw bytes back to host or formating at Pico? might be able to run the simple example logic analyser at 400Mhz and feed bytes into your LogicAnalyzer program.

anyway, I'll test the V2. good work.

startcapture, 2024-09-19:
Do you have a Linux arm64 version of V2?

gusmanb, 2024-09-19:
> Thanks, I will include this for the search engines, and to help raise awareness.
>
> I used the pico logic analyser first time last weekend for a dead Asrock Z170 motherboard, BIOS chip corrupt. [...]

- The packages that I distribute are self-contained, uncompress whatever is in them and that's all you need.
- Ok, now I see what you mean. That's not going to work... In that code there is no real trigger, they start the PIO without any kind of control and set a DMA with a fixed transfer size and let it run, it is not reacting to an event and that's why it uses a single instruction, in fact the PIO program is a single instruction. If you do that is impossible to know when to start capturing properly... In any case I have an idea that might work, will check this weekend if it's possible to do something.

Regarding the ARM64 version, I have uploaded it in the discussion sections.

Cheers.

startcapture, 2024-09-21:
Looking again at example logic analyser in SDK, and man, the PIO is hard to understand. 

it does have a trigger system using WAIT 0 or 1, gpio number. instruction directly injected into the PIO, that will wait on one GPIO only, for a change of state. For example if you included it in your project:

1.  there would be no pre-trigger sampling
2.  can only trigger off one GPIO
3.  can be any GPIO assigned to PIO
4.  can run at 400Mhz sample rate

interestingly WAIT 0 or 1, pin number. is 2 clock cycles. WAIT 0 or 1, GPIO number. is 1 clock cycle.
you can inject a WAIT instruction into PIO anytime, it will be immediate.

for example monitoring a SPI chip select line
initial trigger WAIT 0 GPIO NO. will trigger on Chip Select low. if immediately inject WAIT 1 GPIO NO. it will stop sampling and start again on Chip Select high.
initial trigger WAIT 0 GPIO NO. will trigger on Chip Select low. if immediately inject WAIT 0 GPIO NO. should have no effect as Chip Select line is already low.

my understanding is Pico can go to 420Mhz(100%), 450Mhz integer only and maybe higher if DMA only.
as always double check this info as I'm no expert.

Anyway some food for thought.

gusmanb, 2024-09-21:
> Looking again at example logic analyser in SDK, and man, the PIO is hard to understand.
>
> it does have a trigger system using WAIT 0 or 1, gpio number. instruction directly injected into the PIO, [...]

Where have you seen that WAIT PIN is 2 cycles? The PIO instruction set is deterministic, each instruction takes always 1 cycle to execute, from the datasheet:

    All PIO instructions execute in one clock cycle.

And there is no special statement about it in the instruction description:

```
3.4.3. WAIT
...
3.4.3.2. Operation

Stall until some condition is met.
Like all stalling instructions (Section 3.2.4), delay cycles begin after the instruction completes. That is, if any delay
cycles are present, they do not begin counting until after the wait condition is met.
    • Polarity:
        ◦ 1: wait for a 1.
        ◦ 0: wait for a 0.
    • Source: what to wait on. Values are:
        ◦ 00: GPIO: System GPIO input selected by Index. This is an absolute GPIO index, and is not affected by the state
machine’s input IO mapping.
        ◦ 01: PIN: Input pin selected by Index. This state machine’s input IO mapping is applied first, and then Index
selects which of the mapped bits to wait on. In other words, the pin is selected by adding Index to the
PINCTRL_IN_BASE configuration, modulo 32.
```

Is that noted anywhere else?

> initial trigger WAIT 0 GPIO NO. will trigger on Chip Select low. if immediately inject WAIT 0 GPIO NO. should have no effect as Chip Select line is already low.

Well, that's not true, it will stall for 1 cycle, the instruction must be executed, what will not add is more cycles after the initial one as the condition is already met.

In any case, if you are thinking about having the burst mode using this, that's not going to happen, the current burst mode rearms the trigger inside the PIO program and that's why it is so fast, one single cycle, with this mode you cannot do that as the PIO program does not know when to stop, you program in the DMA how many transfers you want, that triggers an interrupt when it has finished and then the CPU finalizes the operation, if you rearm the PIO and the DMA from the CPU the delay is going to be massive (in comparison to the sampling speed, of course), in the order of 500-800ns being optimistic, the interrupt needs to be propagated, the stack frame needs to be created, the IRQ needs to be acknowledged, the DMA needs to be reconfigured and the PIO restarted.

Anyways, I got it already implemented and not using WAITs but using JMP PIN and the INOVER of the pin which is later flipped.
Here are the links:

[Software](https://mega.nz/file/XCx1AaII#L6gYhgCwwXqRzvHKEftyUNiGIDGrfoaTg67dbY4AZ-g)
[Firmware](https://mega.nz/file/uXQTnCzL#wOJkjqWnWZRToNwpgB6Eto8K1QIYEkEt9vRskBU5q3E)

To enable this mode in the software select the edge trigger and check the "Blast mode" checkbox, it will disable unsupported options and will fix the frequency to the maximum that the device supports.

I have done multiple tests and there are some considerations to have:

- You will capture from the next cycle after the event happens, what I mean is, if you inject a 400Mhz changing signal (for exaple the two phases of a 200Mhz clock) and set the trigger edge to low the first sample will contain the high edge as the JMP will take one instruction.
- You cannot use the level shifters with a 400Mhz signal as these are meant up to 200Mbps so they are out of specs.
- Sampling a 400Mhz signal gives not perfect results, I have tested it injecting a 200Mhz clock and half of the time what is read is always one of the phases, I'm not sure if this is because of the skew of the pico's clock, because the internal capacitance of the pads or the internal propagation, injecting a 100Mhz clock the data is read perfectly seeing each time two samples for each phase of the clock.

I'm going to leave the issue open, when you test it add any findings or problems here please 😄 

Cheers.

startcapture, 2024-09-21:
Where have you seen that WAIT PIN is 2 cycles?

00: GPIO: System GPIO input selected by Index. This is an absolute GPIO index, and is not affected by the state
machine’s input IO mapping.                                     (direct GPIO access, 1 cycle)

01: PIN: Input pin selected by Index. This state machine’s input IO mapping is applied first, and then Index
selects which of the mapped bits to wait on. In other words, the pin is selected by adding Index to the
PINCTRL_IN_BASE configuration, modulo 32.          (indirect GPIO access calculation,  2 cycle)

I found a pointer to the above via a Logic Analyser snapshot on the internet somewhere, looking for information.

Well, that's not true, it will stall for 1 cycle,                   Yes. so would SDK example miss the first sample?

No, I'm not thinking of making Burst Mode or any thing yet, I only just setup Pico SDK, cmake & PIO all new to me.

Interesting, one thing I did think of is maybe:                Get the trigger point close (not exact) and back track in software
OK thanks I will test it all out.

startcapture, 2024-09-21:
oops I forgot to mention I'm on Linux arm64

gusmanb, 2024-09-21:
> oops I forgot to mention I'm on Linux arm64

Ok, here it is for Linux: https://mega.nz/file/CSI00ZCA#HchVmZetGmqh-aQPOCrfQHR_65xnsUVFEcSTPs-DyoA

gusmanb, 2024-09-21:
> I found a pointer to the above via a Logic Analyser snapshot on the internet somewhere, looking for information.

I don't think that statement is true, input mapping in the PIO is done through hardware, there is a mapper between the PIO and the GPIOs so it is not really calculated, it is configured, maybe they refered to their code or something else.

> Well, that's not true, it will stall for 1 cycle, Yes. so would SDK example miss the first sample?

Yes.

> Interesting, one thing I did think of is maybe: Get the trigger point close (not exact) and back track in software OK thanks I will test it all out.

What do you mean with "backtrack"? You don't have samples before the trigger event happens so you can't know what happened before it. That's why on the regular mode you have the presamples and postsamples, to be able to have some info on what happened before the trigger happens.

Cheers.

startcapture, 2024-09-21:
ARM64 Linux, download file is x86 Linux.

site's mentioning PIO WAIT
https://forums.raspberrypi.com/viewtopic.php?t=321852
SEARCH FOR in above
wait pin` being 1 cycle slower than `wait gpio

LOGIC ANALYSER IMAGES
https://forums.raspberrypi.com/viewtopic.php?t=321852

Although you could be correct. after I saw these sites I looked in datasheet and assumed the calculation added 1 clock cycle.

Get the trigger point close (not exact) and back track in software
some questions: when PIO-1 triggers(fast_capture) and sets GPIO-1/GPIO-2 what happens in PIO-0?
does it log some sort of counter, I can see the end of sample calculations but I'm missing some thing.

gusmanb, 2024-09-21:
Trigger PIO is only used by the fast/complex triggers and adds more delay, in my firmware as the main capture PIO is running indefinitely in a ring buffer I can take into account the delay and adjust the trigger point, with the "just capture" that you want to do you cannot do nothing like that, it will trigger once the condition happens and that's all...

startcapture, 2024-09-21:
I'm on ARM64 Linux, download file is x86 Linux, incorrect file

gusmanb, 2024-09-21:
https://mega.nz/file/vLhHGBxa#c_YskBHiExaZmObscPytIJSKN_2HXiBDxXuM5vnLfcw

startcapture, 2024-09-21:
Yep got it all, thanks. I'll set it up tomorrow and give you some feed back.

I have read the datasheet and PIO has counters but are they 32bit (up to 4Gig) or are they count to 0->32? are these what
Logic Analyzer uses to track sample bits? and used in final calculations

Anyway I'm signing out for today
Cheers

startcapture, 2024-09-22:
WOW, I AM TOTALLY IMPRESSED by your work, this is excellent.
This would be the most valuable resource for delving into the Raspberry Pi ecosystem by far.
Big congratulations, gone from 100Mhz sample rate to 200Mhz then 400Mhz in one week.

I have only tested it with PI 4B serial comms UART and SPI, it's super accurate. I haven't done any
propagation delay test yet but I'm totally confident of the 2.5 ns sample time.

one suggestion. on Blast mode if you know what your testing, example UART, can we fill the first two
samples in manually so the measurements are correct. or I suppose it can be saved and edited in text editor.

Re Pico V2, looking online they seem to have some test results for GPIO problem. recommending a 8.2K or less resistor
to GND. Using the Pi & Pico GPIO's  at ~40ohm output impedance, should not be a problem with latching high and getting
stuck. The only problem I can see, if the input to PICO2 GPIO is high (eg. UART in idle), it will leak up to 0.3 ma through
the pin to GND, I suppose with lots of pins this might add up.

You have done some Pico V2 tests already, does this apply to what you have seen? they have ramped it up to 660Mhz, this
would have 1.5ns sample time, can only hope.

Anyway, again GREAT WORK

startcapture, 2024-09-22:
I see in the open device menu, UART ports /dev/ttyAMA0 & /dev/ttyAMA2, actually the Pi 4B has 5 high speed UARTS,
is SERIAL going to be a possibility. if so "you beauty"

gusmanb, 2024-09-22:
> WOW, I AM TOTALLY IMPRESSED by your work, this is excellent. This would be the most valuable resource for delving into the Raspberry Pi ecosystem by far. Big congratulations, gone from 100Mhz sample [...]

Really glad it worked well :)

About filling, well, that makes sense for your use case right now, but I can't know what users may be sampling, so is not something that I could do. If you want to fix the capture you can insert samples [using the edit feature](https://github.com/gusmanb/logicanalyzer/wiki/06---The-LogicAnalyzer-program#editing-captures).
Beware that I'm now fixing this functionality in the new software, if you're going to use it tell me and I will drop a build here when it is completed.

Regarding the UART, I read from the system the available serial ports and use the one you select, if you use a USB to UART adapter and choose one serial port it will work, if there is any missing port I only can assume that it is not enabled in the device. 
In the firmware itself is not possible to expose the UART, every GPIO is used and there is none left for the UART.

Cheers.

startcapture, 2024-09-22:
Yes I will be interested in editing the sample, only want to add 2 clock cycles worth of samples at the beginning of a BLAST mode, can it be input with the pin triggers in the setup section . I am still learning the software capabilities ect. so I am not sure if it is easy, seems to rely on "cut n paste" for insertion, but I may be wrong.

Just curious, I see you have a [24 channel per Pico x 4 Pico's] = 96 channel option, what hardware is being worked on to need
this many channels? subtle hint to get you to forgo 2 pins for UART use. A Raspberry 4B could run 5 logic analysers at once,
although not hardware triggered.

Anyway I'm still BUZZING with how good this is.

gusmanb, 2024-09-22:
> Yes I will be interested in editing the sample, only want to add 2 clock cycles worth of samples at the beginning of a BLAST mode, can it be input with the pin triggers in the setup section . I am [...]

- No, you don't need to copy/paste, read the documentation that I linked, you right-click on the sample marker where you want to insert samples and the composer window will appear, there you can enter a script per channel to create the samples you need, the software includes a full fledged language to create samples, you can emulate any protocol with it.

- It is up to 5x24 = 120 channels 😉. As far as I know users have only reached 48 channels to capture a 24 bit addres plus 8 data lines and control lines, a chain of two, but adding more costed me zero in the software and the hardware is not limited as it is a chain (maybe by signal loses/delays? not sure, never went beyond 5 devices), so, why not to allow it? 😄 

- Regarding the UART, again, there are no free pins at all, the pico only exposes 26 GPIOs, 24 are used for channels and two for the fast/complex triggers, so there is no possibility. If you want to use it through real UART then simply add a USB to UART adapter, they cost less than 1€ 😃

- Once I complete the insert feature in the new software will drop here a new version.

Cheers.

startcapture, 2024-09-26:
Hi, the more I look around, the more I realise how much work has gone into your project. WOW just WOW.
just to clarify, is it 132Mhz for complex_capture, 200Mhz for fast_capture, 400Mhz for blast_capture?

I made a mistake thinking 2 clock cycles missing on blast_capture trigger but it's only 1 clock cycle. You said you were using
JMP PIN and the INOVER, can I see that code anywhere? I've been reading the datasheet PIO and I can now understand
the jump table for trigger and inverting the logic "1" input pins to only trigger on 00000(start of table), ingenious. 

The only way I can see to have 400Mhz sample rate and some pre trigger samples is to have one of the DMA channels spinning
for the pre, and another DMA channel not spinning for blast type trigger for the post.

gusmanb, 2024-09-26:
> Hi, the more I look around, the more I realise how much work has gone into your project. WOW just WOW. just to clarify, is it 132Mhz for complex_capture, 200Mhz for fast_capture, 400Mhz for [...]

-Those are the trigger speeds, fast/complex/edge capture up to 200Mhz but the complex trigger as it uses three instructions to detect the condition sees its effective speed reduced to Fck/3, in this case 133Mhz. Basically meanwhile the trigger condition is not faster than 133Mhz it will be detected properly and the software will correct the trigger moment so it will be transparent (can have up to +-1 sample of difference, but as you have pretrigger samples, it should be easy to detect this situation).

-INOVER is set on the blast capture start routine and JMP PIN in the blast capture PIO program, the iNOVER is simply to use the JMP PIN and be able to have positive edge or negative edge triggers.

-About the presamples, no idea how you want to do it, if you have a DMA ring buffer (like I do in the other modes) for the pretrigger, how are you going to stop the DMA or the PIO at the moment of the trigger? the main PIO machine cannot raise anything to stop the DMA at the correct moment (no instructions left, and even if they are, an IRQ takes time to be propagated) and without an stop it will continue sampling indefinitelly. 
In the other modes I pass a counter to the PIO machine that is decremented on each sample and when it reaches zero the PIO stops pushing data to the FIFO and raises an IRQ to the CPU, in this way the DMA starves and gets blocked and the CPU is able to stop the DMA and know exactly in which position of the ring buffer the capture ended.

Cheers.

startcapture, 2024-09-27:
Thanks for that info, so effectively up to 200Mhz(pre sample and multiple trigger types) or 400Mhz(no pre sample, single pin trigger).
INOVER different to what I was thinking, are you using same jump table as in V6.0?

[YOU SAID: About the presamples, no idea how you want to do it], No, just floating idea's that might work,

I can't really figure from the datasheet. if you turn off autopull after triggering, will that stall the ISR->FIFO transfer even if
FIFO is emptied via DMA? an IRQ0, would you have an estimation on how long would that take just to stop, but not process
a DMA channel?

startcapture, 2024-09-28:
Again for the search engines, and to help raise awareness. 400Mhz logic analyser.

suspected the data sheets for the 74HC.. series IC's were not updated to reflect the manufacturers process upgrades
and proposed to use 74HC244 driver if the propagation delay was ok.  Datasheet ~70ns. logic analyser @400Mhz, 17.5ns,
(I was hoping for 5ns, wishful thinking) so it PAYS TO CHECK.

NGSPICE electrical simulation program, using a Process Design Kit (Skywater ect.) I modeled a GPIO with load specs 20ma
and with logic analyser, use unloaded GPIO as trigger and known load on another GPIO to correct the model by checking 0(low)-2.0(high) volt delay against the simulation.

and Logic Analyser does not load the circuit as much as a scope probe.

Looking forward to Pico2 version

closed 2024-10-07 by gusmanb