gusmanb/logicanalyzer #145: Color Picker anomalous reactivation in LogicAnalyzer v6.0

NobodyYouKnow, 2024-09-18:
In LogicAnalyzer v6.0, once Capture Settings -> Color Picker is activated and then closed (or Canceled), any subsequent click in the Capture Settings window will re-activate the Color Picker. This occurs in the Linux beta distribution. Issue can be cleared by closing the Capture Settings window.

gusmanb, 2024-09-18:
Please, do not open bugs about V6.0 here, add them to the thread as it is stated in the thread where you downloaded the build.

closed 2024-09-18 by gusmanb

NobodyYouKnow, 2024-09-18:
Roger that. My apologies. Somehow I missed that.

On Wed, Sep 18, 2024 at 2:48 PM Agustín Gimenez Bernad  <

> [image: IronVest]
>
> You can *reply directly* and still *remain anonymous.*
>
> To protect your privacy, do not forward this message, or add new
> recipients like CCs or BCCs.
>
> Is this email spam or just don’t want to see it anymore? Disable
> Forwarding
>
> Please, do not open bugs about V6.0 here, add them to the thread as it is
> stated in the thread where you downloaded the build.

gusmanb, 2024-09-19:
> Roger that. My apologies. Somehow I missed that.
> […](#)
> -- R/
> -Pat

No worries 😄 
As this is still in development and I may upload multiple versions I prefer to have all these reports in a single thread so if anyone is going to download it they can find the latest version.

Cheers.