gusmanb/logicanalyzer #155: Edge trigger fired immediately

cavermanus, 2024-10-23:
Version: 5.1.0.0 
Version: 6.0 beta 3
OS: Windows 10

Hi!

The positive edge trigger fired immediatelly, if the selected trigger channel level HIGH, when the capture started. 
The negative edge trigger fired immediatelly, if the selected trigger channel level LOW, when the capture started.

Example situation:
The device to be measured not powered on yet, so all lines 0. I'd like to watch the changes when selected line firstly goes low after the power on the device. The trigger fired before I can powering the device.

For issue check I tied the trigger channel to GND, select negative edge and ACCEPT. The trigger fired.
An other check I tied the trigger channel to GND, select negative edge with burst mode and count e.g. 2 and ACCEPT. The trigger fired two times.

It's same for positive edge, but need to tie to 3V3.

gusmanb, 2024-10-23:
This is how it is designed.

closed 2024-10-23 by gusmanb

cavermanus, 2024-10-23:
what is the advantage of this solution?

gusmanb, 2024-10-23:
> what is the advantage of this solution?

It is fast, can check for a trigger event with a single PIO instruction

cavermanus, 2024-10-24:
Are you planning a burst mode for pattern trigger?

gusmanb, 2024-10-24:
> Are you planning a burst mode for pattern trigger?

Not for now, the problem with the pattern trigger is that the trigger program runs in a different PIO unit (it uses all the available instructions) and you cannot intercomunicate two PIO units except without a "cheat", using IO pins, that's what I did to intercomunicate Trigger->Capture, and to implement the burst mode I would need to use more pins to do the Capture->Trigger communication, that are not available, or to swap the GPIO directions what would consume capture cycles and I'm not sure how well that would work...

Maybe I could go Capture->CPU->Trigger but that will add delay for sure and the rearm of the trigger would not be as fast as in the current burst mode (it takes only two capture cycles right now).