gusmanb/logicanalyzer #165: Compatibility with after-market pi picos (for type-C)

dzervas, 2024-12-12:
I see that the daughter board has length-matching traces so I wanted to ask if after market pi picos are compatible (I just want type C). For example [this](https://www.aliexpress.com/item/1005008143232437.html) one.

How critical are the length-matched traces? Is there a way to "calibrate" based on the new board?

gusmanb, 2024-12-12:
Trace length should not be critical, I designed the board to have a maximum skew of 100 picoseconds, so unless the board adds more than 1 nanosecond (what is nearly impossible due to its size) it should not affect the results.

mengstr, 2024-12-12:
Hehe..  yes,  not just only "nearly impossible" but I rather say "quite impossible" considering that in 1 ns a signal propagates like 160mm (half a foot) on a FR4 PCB. 😆 

Luckily us mere mortals that don't design PCBs having DDR5 on them usually don't have to worry about stuff like that. But it would be fun to try doing a design like that once in the life though.

gusmanb, 2024-12-12:
A final warning, ensure the pinout is the same, I have some pico clones (pico 1, not 2) that have a slightly different pinout and expose one more GPIO, beware as it could cause a short if your clone has someting similar.

closed 2024-12-12 by gusmanb